gnuradio/gnuradio #6613: Agc constructor sets max gain variable to 65536

cmrincon, 2023-03-26:
### What happened?

AGC, AGC2 and AGC3 constructors sets max gain to 65536, so variable in the yml file is useless.

### System Information

OS: all
GR: source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-386-g379f5a3e

### Steps to Reproduce the Problem

Check agc3_xx_impl.cc, agc2_xx_impl.cc and agc_xx_impl.cc

willcode, 2023-03-27:
The yml creates the block, at which point max_gain is set by the constructor. Then it does a set_max_gain() which changes the value. Is that sufficient?

As a side note, it's odd that the constructor calls the setters instead of setting the values. That might be to get test_and_log_value_domain() run. Someone must have had a good reason.

cmrincon, 2023-03-27:
Also note that apparently, agc and agc2 blocks does nothing but set the values and the alignment. At first glance values can be safely set at constructor. 

I would love to hear the reason to do such odd things.

marcusmueller, 2023-04-08:
@cmrincon That's possibly a parameter that was added later on. In the sense of API stability, it wasn't made part of the constructor, maybe. We can't tell you – I do agree the API is now not consistent, but since there's no technical downside setting it after construction, I'm not calling it a bug.

Now, generally, I'd say "oh yeah, you're right, this lacks beauty; we'd very much welcome a pull request to add that parameter as last optional parameter to the `make` functions and constructors". Would you willing to do that?

closed 2023-04-08 by marcusmueller