gnuradio/gnuradio #6614: GNU Radio v3.10.5.1 RTL-SDR Source return code -11

rfrogman, 2023-03-31:
### What happened?

Upgraded to version v3.10.5.1 from 3.10.1.1 to fix issue with QT GUI Compass only rendering a vertical line. Using the information on GNU Radio page:

    sudo add-apt-repository ppa:gnuradio/gnuradio-releases
    sudo apt-get update
    sudo apt-get install gnuradio python3-packaging

Now the I receive >>> Done (return code -11) from the RTL-SDR Source block. For troubleshooting purposes I used a simple FM radio I had working in the v3.10.1.1 version and it produced the same error.

### System Information

OS: Linux Distro BackBox 8 (built on Ubuntu Jammy 22.04.2LTS)
GR Installation Method: 
 v3.10.1.1 
 sudo apt-get install gnuradio
 
Upgrade to v3.10.5.1
    sudo add-apt-repository ppa:gnuradio/gnuradio-releases
    sudo apt-get update
    sudo apt-get install gnuradio python3-packaging

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.1

### Steps to Reproduce the Problem

[image]

### Relevant log output

```shell
Generating: '/home/rfrogman/RTL2832Utest.py'

Executing: /usr/bin/python3 -u /home/rfrogman/RTL2832Utest.py

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
Found Rafael Micro R820T tuner
libusb: warning [libusb_exit] device 4.1 still referenced
libusb: warning [libusb_exit] device 3.2 still referenced
libusb: warning [libusb_exit] device 3.4 still referenced
libusb: warning [libusb_exit] device 3.3 still referenced
libusb: warning [libusb_exit] device 3.5 still referenced
libusb: warning [libusb_exit] device 3.1 still referenced
libusb: warning [libusb_exit] device 2.1 still referenced
libusb: warning [libusb_exit] device 1.1 still referenced
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!

>>> Done (return code -11)
```

rfrogman, 2023-03-31:
I just test the same .grc files on a VM of DragonOS which is on the same version of QT but v3.10.4.0 GNU Radio and they work.
Qt version 5.15.3

marcusmueller, 2023-04-08:
This is not an error within GNU Radio – it's in gr-osmosdr, which is not part of GNU Radio. 

(in case you haven't done that: after changing GNU Radio versions, you'll need to rebuild gr-osmosdr).

What we can definitely offer: Instead of using the (gr-osmosdr-supplied) "RTL-SDR Source" try the GNU Radio-shipped "Soapy RTLSDR Source".

closed 2023-04-08 by marcusmueller