gnuradio/gnuradio #6615: GRC: Parser does not support using attributes or methods of variables in callback parameters

Notou, 2023-04-03:
### What happened?

When a GRC variable (from a Variable block, a QT range, Filter, Constellation, ...) is an object, one might want to call for attributes or methods of said object, and use that in a parameter of another block.

This works at initialisation but not at runtime. When a block parameter (that can handle callbacks) is set using a variable method, no callback is created in the python file. And if the method is used alongside another variable, the callback is created but the method is not prefixed with `self`.

It looks like the parser tasked with searching the parameter strings for variable names considers `variable.method` as a whole word and thus doesn't see it as a known name.

I believe [this function](https://github.com/gnuradio/gnuradio/blob/4f3a0b82c54c5dfeabdd6278d69e46302261aff4/grc/core/utils/expr_utils.py#L14) is related to that issue.

### System Information

OS: Linux mint 21.1
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.1

### Steps to Reproduce the Problem

Create a constellation variable with `digital.qam_constellation(2**bps)` and a QT Range to define `bps`.
This will allow to change the type of QAM (QPSK, 16QAM, ...) at runtime.

Use that constellation variable in a Chunks to Symbols block with `Symbol table: constellation.points()`.
Generate a bit stream to feed that block and observe the generated constellation.

The Constellation doesn't change when moving the slider.
In the generated python file, there is no callback corresponding to the Chunks to Symbols block

Attached a .grc to reproduce: [param_parser_example.grc.zip](https://github.com/gnuradio/gnuradio/files/11136602/param_parser_example.grc.zip)

Jyngle, 2023-04-25:
Hello Notou, 
I have encountered the same issue as you. The problem is indeed in the 'expr_replace' function, specifically in the line: 
"expr_splits = _expr_split(expr, var_chars=VAR_CHARS + '.')"

The bug was created while trying to fix another one in the commit '1319af2', which corresponds to adding '.' to 'VAR_CHARS'. I have made a correction that works for me. It's not very elegant, and I haven't had much time to thoroughly investigate the issue, but it does the job. However, I advise using it with caution as I haven't extensively tested it.

Add code in **expr_utils.py**:

      def _modified_expr_split_(expr, var_chars=VAR_CHARS):
        toks = list()
        tok = ''
        quote = ''
    
        for char in expr:
            if quote or char in var_chars:
                if char == quote:
                    quote = ''
                tok += char
            elif char in ("'", '"'):
                toks.append(tok)
                tok = char
                quote = char
            else:
                toks.append(tok)
                if char in ("("):
                    var_chars = var_chars.replace('.', '')
    
                if char in (".", ")", "'", '"') and ('.' not in var_chars):
                    var_chars = var_chars + "."
    
                if char in ("."):
                    tok = char
                    continue
    
                toks.append(char)
                tok = ''
    
        toks.append(tok)
        return [t for t in toks if t]

      def modified_expr_replace(expr, replace_dict):
          """
          Search for vars in the expression and add the prepend.
      
          Args:
              expr: an expression string
              replace_dict: a dict of find:replace
      
          Returns:
              a new expression with the prepend
          """
          expr_splits = _modified_expr_split_(expr, var_chars=VAR_CHARS + '.')
          for i, es in enumerate(expr_splits):
              if es in list(replace_dict.keys()):
                  expr_splits[i] = replace_dict[es]
          return ''.join(expr_splits)

Modified the **_callbacks(** function in **top_block.py** :

        callbacks_all = []
        for block in fg.iter_enabled_blocks():
            callbacks_all.extend(expr_utils.modified_expr_replace(
                cb, replace_dict) for cb in block.get_callbacks())

willcode, 2023-06-09:
(ignore that mention - linked to wrong thing)