gnuradio/gnuradio #6625: Improper Explanation of Differential Phasor

DevanshRaizada, 2023-04-09:
### Feature Description

Not really a bug, but the definition of Differential Phasor in 'Category: Block Docs' is short and unclear and doesn't have an example to explain it. 
[Link](https://wiki.gnuradio.org/index.php?title=Differential_Phasor)

### Feature Urgency

medium (would be nice to have in the near future)

marcusmueller, 2023-04-09:
Hi @DevanshRaizada! Seeing that's a wiki page: do you feel up to the task to improving that documentation?

DevanshRaizada, 2023-04-10:
@marcusmueller Yes, surely I would love to improve that documentation.

marcusmueller, 2023-04-10:
Ha, wonderful! @DevanshRaizada then please do that: on the wiki page you've linked to, there's an "Edit" link (somewhere on the top right). Unless you have a wiki account already, you can make an edit, and then one of the moderator wiki users has to click on "this edit is OK". Then we'll have fixed the issue in no time!

DevanshRaizada, 2023-04-10:
@marcusmueller Thank you for th guidance.

DevanshRaizada, 2023-04-13:
I have updated the page and now I am closing the issue.

closed 2023-04-13 by DevanshRaizada

marcusmueller, 2023-04-13:
Thank you so much, @DevanshRaizada !

DevanshRaizada, 2023-04-13:
@marcusmueller No need I'm actually looking for meaningful ways of contributing.