gnuradio/gnuradio #6638: Vector accumulator block

cdekker53, 2023-04-11:
### Feature Description

I am looking for an embedded Python Sync block 'Accumulator' , that accepts an input vector and adds it to the last output vector. In my application it is for finding a pulse with a fixed and known time interval, but deeply burried in the noise.
I seems a simple functionality but somehow I don't succeed, starting with the standard Embedded Python Block script:

import numpy as np
from gnuradio import gr

class blk(gr.sync_block):  
    def __init__(self, gain=1.0, vectorSize=13462):  
        gr.sync_block.__init__(
            self,
            name='Accumulator',   # will show up in GRC
            in_sig=[(np.complex64,vectorSize)],
            out_sig=[(np.complex64,vectorSize)]
        )
        self.accum_items=[(np.complex64,vectorSize)]

    def work(self, input_items, output_items):
        self.accum_items =  input_items       # + self.accum_items
        output_items[0][:] =   self.accum_items[0]
        return(len(output_items))

The output goes via a vectot to serial block to a QT Time sink block with 13462 Number of Point (13462  is just an example)
When I remove the # character, the output remains 'empty'.

### Feature Urgency

high (impacts broad portion of ecosystem)

### More Information

In Simulink this was easy

cdekker53, 2023-04-11:
Next I give a retry of the text and source code which is badly formatted:

As I can't find a GNURadio block 'Accumulator', I tried to mimic with an embedded Python Sync block, which accepts an input vector and adds it to the last output vector (of similar length). In my application it is for finding a pulse with a fixed and known time interval, but deeply burried in the noise.
It seems a simple functionality but somehow I don't succeed, starting with the standard Embedded Python Block script below (vectorSize=13462 is just an example):

import numpy as np
from gnuradio import gr

class blk(gr.sync_block): 
    def __init__(self, gain=1.0, vectorSize=13462):  
        gr.sync_block.__init__(
            self,
            name='Accumulator',   
            in_sig=[(np.complex64,vectorSize)],
            out_sig=[(np.complex64,vectorSize)]
        )
        self.accum_items=[(np.complex64,vectorSize)]

    def work(self, input_items, output_items):
        self.accum_items =  input_items  # + self.accum_items
        output_items[0][:] =   self.accum_items[0]
        return(len(output_items))

The output of the above is identical to the input vector. The output goes via a Vector to Serial block to a QT Time Sink block with 13462 as its Number of Points parameter.
When I remove the # character the output remains empty or 0, and the accumulation doesn't seem to work, whatever I try.

My question is: does a GNURadio block exists already for this Accumulator functionality? And what am I doing wrong?

Notou, 2023-04-12:
The [moving average](https://wiki.gnuradio.org/index.php/Moving_Average) seems to the block you are looking for.
It does an element wise moving sum over the input vectors, that you can scale (or not) by the amount you want

cdekker53, 2023-04-13:
I am not sure if this it what I am looking for.  Please let me explain.

I am receiving a continuous stream of K-samples per second from space. It seems to be only (Gaussian) noise, but I know that every N-samples a (very weak) pulse is received in it (e.g. from a pulsar) and I want to find out a which moment. If I would use a Moving Average model as I understand it, this pulse is averaged out beween all the noisy samples and thus I don't see anything.

Therefore my intended approach is to first convert the input with a 'Stream to Vector block' output to a sequence of vectors with a vector length equal to  the above N-samples, Next I  want to add these vectors of equal length N to one another. The accumulated resulting vector, shows averaged-out noise-only samples, but at the repeating pulse sample moments the output will increase above this output noise floor and I should see the exact (sample) moments when the pulses are received. In the QT Time Sink - with exactly N-samples horizontal) I can see that output noise floor with the increasing pulse output at a steady moment (e.g. at sample moment X on the QT Time Sink display).

I called this the intended"Vector Accumulator".  Hopefully this is a more understandable  explanation?

Notou, 2023-04-13:
Perhaps I could have been clearer myself.

The moving average block can take vectors as input.
In that case, it will work separately on the vectors elements:
Say that you have a N length input, the vector output at time $t$ will be $[o^t_0, o^t_1, ..., o^t_n, ... o^t_N]$
And $o^t_n = (i^t_n + i^{t-1}_n + ... + i^{t-L}_n) * scaling$  (with $i^t$ the input vector at time $t$)
So each time element of your N samples period will be averaged over the L previous time elements at the same position in the period

You can choose L however you want (big values might raise the memory consumption) and also the scaling.
You can even set scaling to 1 to just do a moving sum.

willcode, 2023-05-10:
What would keep the accumulator bounded in such a block? Would it periodically reset?