gnuradio/gnuradio #6646: About the Polyphase Clock Sync Block

saro9e, 2023-04-30:
Dear all,
I have a few questions about the Polyphase Clock Sync Block in reference to https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/pfb_clock_sync_ccf_impl.cc

line no 67:    d_damping = 2 * d_nfilters;   -->  d_nfilters=32, should this not be around 0.707

line no 366:  based on the 'time_est' tag, determine the filter index, d_k. But, this completely bypasses the TED and the control loop as in line no 374 d_filtnum = (int)floor(d_k);
then why do we need this Polyphase Clock Sync Block at all?

How does the main loop take care of which input samples are selected for the determination of d_error and d_k such that all the input samples come from the same input symbol?
With thanks and regards 
Saurav

willcode, 2023-04-30:
(1) is probably a bug, since
```
    // Set the damping factor for a critically damped system
    d_damping = 2 * d_nfilters;
```
is not consistent with
```
void pfb_clock_sync_ccf_impl::set_damping_factor(float df)
{
    if (df < 0 || df > 1.0) {
        throw std::out_of_range(
            "pfb_clock_sync_ccf: invalid damping factor. Must be in [0,1].");
    }

    d_damping = df;
    update_gains();
}
```

saro9e, 2023-04-30:
Thanks @willcode. Can anyone please shed light on points 2 and 3?

marcusmueller, 2024-02-13:
Closing this: Extracted the bug to 7130, and the other two issues are questions; for questions, please refer to the mailing list or our chat server.

closed 2024-02-13 by marcusmueller