gnuradio/gnuradio #6648: [MSVC] 'sscanf()' and "%m"

gvanem, 2023-05-04:
### Feature Description

Building **gnuradio-qtgui.dll** on MSVC (with OpenGL support), I noted this important warning:
```c
QFosphorColorMapper.cc(67): warning C4476: 'sscanf' : unknown type field character 'm' in format specifier
QFosphorColorMapper.cc(67): warning C4474: 'sscanf' : too many arguments passed for format string
QFosphorColorMapper.cc(67): note: placeholders and their parameters expect 1 variadic arguments, but 2 were provided
```

A support for `%m` seems Linux specific too me. Ref:  http://linux.die.net/man/3/scanf 

No word on this on [MSDN](https://learn.microsoft.com/en-us/cpp/c-runtime-library/scanf-type-field-characters?view=msvc-170). Not sure if MinGW-w64 supports this. 

For MSVC, that code will crash at runtime; `free()` of an illegal pointer.

I tried this feeble attempt instead:
```diff
--- a/gr-qtgui/lib/QFosphorColorMapper.cc 2023-04-26 15:19:11
+++ b/gr-qtgui/lib/QFosphorColorMapper.cc 2023-05-04 13:47:35
@@ -64,7 +64,13 @@

         /* Are we in a palette ? */
         if (kp < 0) {
+#ifdef _WIN32
+            char buf[41] = { '\0' };
+            name = buf;
+            sscanf(line.constData(), "%d %40s", &kp, buf);
+#else
             sscanf(line.constData(), "%d %ms", &kp, &name);
+#endif

             if (kp)
                 gradient = new QLinearGradient();
@@ -75,8 +81,10 @@

             addPalette(name, pixmap);

+#ifdef _WIN32
             /* Release name & prepare for next */
             free(name);
+#endif
             name = NULL;

             kp = -1;
@@ -96,7 +104,9 @@
                 delete gradient;
                 gradient = NULL;

+#ifdef _WIN32
                 free(name);
+#endif
                 name = NULL;

                 kp = -1;
```

Must be a better way.

### Feature Urgency

high (impacts broad portion of ecosystem)

marcusmueller, 2023-05-18:
Elevating this to bug status. Yeah, also, why are we doing `sscanf` in a C++ program…

willcode, 2023-05-19:
Related to #6515 
@mbr0wn @gmeyerho can we replace the `sscanf()` functions with a safe alternative?

willcode, 2023-05-19:
Keep cool!

marcusmueller, 2023-05-23:
@gvanem do you think #6676 works for you?

PR #6676 (2023-06-01): qtgui/QFosphorColorMapper: std::sstream+string instead of sscanf("%m")+char*
Fixes #6648

(hopefully)

## Description

#6648 highlights that we introduced GCC extension of the C sscanf to the codebase. That `%ms` is handy when dealing with unknown length string data in C: it avoids having to scan for string length first, then `malloc` a buffer, then do the actual `scanf`. But: That problem is a C-exclusive problem. C++ has `std::string`, and `std::stringstream` (though the latter *could* be nicer).

So, we replace the usage of sscanf and raw C strings with `std::string` altogether and hopefull [... cut]
closed 2023-06-01 by willcode