gnuradio/gnuradio #6652: Soapy SDRPlay Source fails with AGC=True

tko, 2023-05-09:
### What happened?

I changed the RF Options / AGC to `True` and the source doesn't start. The warning and error in log output do not show with AGC=False

Expected automatic gain to be enabled along with a new parameter to choose a level, AFAIK there are parameters to adjust the AGC.

### System Information

OS: Mac
GR Installation Method: homebrew

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.6.0

### Steps to Reproduce the Problem

1. Add Soapy SDRPlay Source
2. Edit Preferences; set Center Freq, and change AGC to `True`
3. Add Null Sink and connect
4. Execute

### Relevant log output

```shell
[INFO] devIdx: 0
[INFO] SerNo: B0002P0001
[INFO] hwVer: 1
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
[WARNING] Not updating IFGR gain because AGC is enabled
[INFO] Using format CF32.
[ERROR] error in activateStream() - Init() failed: sdrplay_api_Fail

SoapySDRUtil -probe show following:

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSP1
  sdrplay_api_api_version=3.070000
  sdrplay_api_hw_version=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=1, type=string, options=(0, 1, 2, 3)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 42] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 3] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.096, 0.125, 0.192, 0.25, ..., 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```

willcode, 2023-05-12:
Replicated, taking a look.

willcode, 2023-05-13:
@tko If you are able, try  out this change. The SDRPlay block gains and AGC were pretty broken.

tko, 2023-05-13:
Didn't work out of the box, default gain needed some tweaking for me
```
[WARNING] Not updating IFGR gain because AGC is enabled
source :error: Gain RFGR out of range: 0 <= gain <= 3
[INFO] Using format CF32.
[ERROR] error in activateStream() - Init() failed: sdrplay_api_Fail
```
I think usable range depends on the actual device so maybe there is no universally good default. The error message is little unhelpful as it doesn't correspond with any of the settings really.

PR #6665 (2023-05-21): soapy: sdrplay agc and gain fixes
## Description
SDRPlay gains are actually attenuations
- grdb: gain reduction of 20 to 59 dB
- lna state: model-specific attenuation steps for LNA

LNA state is independent of AGC. Gain reduction is N/A when AGC is on.

SoapySDR attempts to do automatic distributions of gains if setGain() is called without a gain name. This is what gr-soapy was trying to do and it resulted in invalid gain values.

AGC has a setpoint, which was not being set. This is probably why a crash was happening when AGC was set.

This change:
- Adds the AGC setpoint
- Separates gain and  [... cut]
closed 2023-05-21 by willcode