gnuradio/gnuradio #6668: Documentation missing for two PLL blocks

tko, 2023-05-17:
### What happened?

Trying to read the docs for PLL Frequency Detector and PLL Carrier Regeneration blocks, finding they're missing.

### System Information

OS: Mac
GR Installation Method: homebrew

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.6.0

### Steps to Reproduce the Problem

1. Add PLL Frequency Detector and PLL Carrier Regeneration blocks
2. Click the "Visit Wiki Page" in Properties / Documentation

marcusmueller, 2023-05-18:
Hi @tko,

formerly

- PLL Carrier regeneration: You're right, there's a forwarding missing there: the Block docs are under https://wiki.gnuradio.org/index.php?title=PLL_Ref_Out
-  same, but different URL

The problem was that someone (cough, me) renamed these blocks in the source code to give them the more intuitive names they have now, but forgot to create forwardings to the old name (or vice versa, forward the old name to the new name).

Fixed that now; could you test it works for you?

closed 2023-05-18 by marcusmueller

tko, 2023-05-18:
Yep, works fine. The flowgraphs on those pages still use the old names. While "PLL Freq Det" is probably close enough to not be too confusing, the "Ref Out" is so different it could use a mention