gnuradio/gnuradio #6675: QFosphorColorMapper leaks d_shader, i.e., a QGLShaderProgram on destruction

marcusmueller, 2023-05-19:
### What happened?

This is probably not a critical leak, but the constructor of QFosphorColorMapper makes a `new` object, and there's no explicit destruction

### System Information

n/A

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-465-gc218dffcb

### Steps to Reproduce the Problem

source code lecture of the .cc file

willcode, 2023-05-19:
Could you put a source code link in so we can find it easily later?

Related to #6515

japm48, 2023-05-20:
> In the constructor:
> 
> https://github.com/gnuradio/gnuradio/blob/c218dffcbac4e8fe84e0bb0e7794ab174c3a4fbb/gr-qtgui/lib/QFosphorColorMapper.cc#L18-L20
> 
> But this class has no destructor, and the pointer is nowhere passed to delete.

`QFosphorColorMapper` inherits from `QObject`, so, if I'm not mistaken, there is a (default) destructor that calls the parent class destructor, and then all its `QObject` children are deleted in turn. AFAIK, this is how the Qt's tricky ownership model works.
Note that `QGLShaderProgram(this)` sets a `QFosphorColorMapper` as its parent.
See: https://forum.qt.io/topic/110945/scope-and-destruction-of-parent-child-model-qobjects

Also, unrelated to this, it looks like `QGLShaderProgram` is getting deprecated, and its replacement is `QOpenGLShaderProgram`.

closed 2023-05-23 by marcusmueller