gnuradio/gnuradio #6696: Double selector usage causes gnuradio scheduler to hang

scrovy, 2023-06-02:
### What happened?

I have a GR flowgraph on Python for which the graphical representation is somewhat as below (the flowgraph was not made on GRC but directly on Python; below was adapted from an older prototype; sorry for the poor paint skills). On the upper branch, the samples can go through one of the 4 algorithms or bypassed, depending on the position of the double "selector" switch -- always same position on left and right hand side. On the lower branch, once in a while some samples are taken and a (classifier) algorithm produces the position of the switch to be used, which is given to the upper branch and printed out on the "message debug" block. The command handler in the upper branch gets the asynchronous messages with the desired switch positions and a handler hooked to that port simply runs
`self.selector1.set_output_index(algo)`
`self.selector2.set_input_index(algo)`
where `selector{1,2}` are the selector objects (rhs and lhs, respectively) and `algo` is the variable resulting from parsing the message.
[image]

So far so good, and it seems to work fairly, but the flowgraph blocks under random circumstances when the switch toggles. This does not happen always and looks like some race condition. No messages are printed and the work functions of the algorithms and the switches are not called anymore. I've copied the `selector` code to an out of tree block so as to enable debugging of my own, and I see that when it blocks, no work function of either selector or any of the algorithms is called. Only the `forecast` function of the right-hand side (multiple input single output) selector is called. The number of output samples passed to it is a geometric sequence of powers of 2 that resets circularly: 4096, 2048, 1024, ..., 4, 2, 1, 4096, 2048, ....
Before switching, everything seems neat with the `general_work` function on the switches only having a non-zero number of `ninput_items` on the stream which is active before switching, and also some non-zero number in `noutput_items` (the smallest of these numbers is taken in `to_copy` as in the current code, and these was confirmed with debug logging). When the classifier toggles the switch, the `set_input_index` and `set_output_index` functions of both selectors are called as expected, and the indices are updated; but after that the `general_work` function is not called anymore (only `forecast` on the right-hand side as mentioned above).

This might be related to this recent issue:
https://github.com/gnuradio/gnuradio/issues/5650

I've run it with the latest changes in `maint-3.10` which seem to address this or a related issue, but the locks still happen. Any ideas?

### System Information

OS: Linux Ubuntu Ubuntu 20.04.5 LTS, Kernel 5.13.0-44-generic
GR Installation Method: Source, from current `maint-3.10`: commit `0ef8fdea4aab85151b83afee602b52dfe2f83d53`

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

v3.10.6.0-29-g0ef8fdea

### Steps to Reproduce the Problem

Difficult to reproduce due to race condition nature. You can try to make a flowgraph with a double selector and some sync block branches in between as depicted above in the upper branch. Asynchronously to that call
`set_input_index(algo)`, `set_output_index(algo)` on the selectors with some branch `algo` once in a while. And hope that it locks. The used sample rate is quite high (20 MHz).

### Relevant log output

```shell
Here a snippet of my debug log of the selectors (general_work, forecast, set_input_index, set_output_index and forecast were being monitored using `d_logger->debug` instructions), and of the sync blocks (work function monitorred with `d_logger->debug`).
`[custom_selector(5)]` was the input and `[custom_selector(6)]`. (The term custom is misleading; the selector was just a copy-paste from maint-3.10, with the extra d_logger->debug instructions).
Variables explanation:
`iidx=0 oidx=0 iitm=384 0 0 0 0 ) oitm=256 cp=256` The general_work function was called, and these variables in order: d_input_index, d_output_index, ninput_items, noutput_items, to_copy (formatting was defective for an unknown reason)
`forecast iidx=0 oidx=0 iitm=128 ) oitm=128` The forecast function was called, similar variables as above except for to_copy which is not present

custom_selector :debug: [custom_selector(5)] check_topology called with ninputs=1, noutputs=5
custom_selector :debug: [custom_selector(6)] check_topology called with ninputs=5, noutputs=1
[...]
custom_selector :debug: [custom_selector(5)] iidx=0 oidx=0 iitm=6785 ) oitm=4096 cp=4096
custom_selector :debug: [custom_selector(6)] forecast iidx=0 oidx=0 iitm=1 0 0 0 0 ) oitm=1
custom_selector :debug: [custom_selector(6)] forecast iidx=0 oidx=0 iitm=4095 0 0 0 0 ) oitm=4095
custom_selector :debug: [custom_selector(6)] forecast iidx=0 oidx=0 iitm=2047 0 0 0 0 ) oitm=2047
custom_selector :debug: [custom_selector(6)] iidx=0 oidx=0 iitm=2048 0 0 0 0 ) oitm=2047 cp=2047
custom_selector :debug: [custom_selector(5)] forecast iidx=0 oidx=0 iitm=4094 ) oitm=4094
custom_selector :debug: [custom_selector(5)] iidx=0 oidx=0 iitm=4685 ) oitm=4094 cp=4094
custom_selector :debug: [custom_selector(6)] forecast iidx=0 oidx=0 iitm=2048 0 0 0 0 ) oitm=2048
custom_selector :debug: [custom_selector(6)] iidx=0 oidx=0 iitm=4097 0 0 0 0 ) oitm=2048 cp=2048
custom_selector :debug: [custom_selector(5)] forecast iidx=0 oidx=0 iitm=2048 ) oitm=2048
custom_selector :debug: [custom_selector(5)] iidx=0 oidx=0 iitm=8191 ) oitm=2048 cp=2048
custom_selector :debug: [custom_selector(5)] set_output_index called
custom_selector :debug: [custom_selector(5)] set_output_index setting to 2
custom_selector :debug: [custom_selector(6)] set_input_index called
custom_selector :debug: [custom_selector(6)] set_input_index setting to 2
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 4096 0 0 ) oitm=4096 # <- blocks here
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 2048 0 0 ) oitm=2048
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 1024 0 0 ) oitm=1024
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 512 0 0 ) oitm=512
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 256 0 0 ) oitm=256
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 128 0 0 ) oitm=128
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 64 0 0 ) oitm=64
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 32 0 0 ) oitm=32
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 16 0 0 ) oitm=16
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 8 0 0 ) oitm=8
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 4 0 0 ) oitm=4
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 2 0 0 ) oitm=2
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 1 0 0 ) oitm=1
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 4096 0 0 ) oitm=4096
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 2048 0 0 ) oitm=2048
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 1024 0 0 ) oitm=1024
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 512 0 0 ) oitm=512
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 256 0 0 ) oitm=256
custom_selector :debug: [custom_selector(6)] forecast iidx=2 oidx=0 iitm=0 0 128 0 0 ) oitm=128
[...]
```

scrovy, 2023-06-04:
If I cannot debug the issue with reasonable effort or hints from here that's indeed what I'm gonna do
P.S.: the sync blocks (algorithms) were done in C++, not Python (the latter was only used for the flowgraph ie hooking up the individual blocks)

willcode, 2023-06-04:
I'm going to close this as an issue, since this is more of a limitation of how GR buffering/scheduling works. Feel free to re-form this as a feature request. Also, bring it up on chat.gnuradio.org if you haven't already, to see if anyone has ideas.

closed 2023-06-04 by willcode