gnuradio/gnuradio #6699: TCP Sink in server mode leaks memory

argilo, 2023-06-02:
To reproduce, instantiate a TCP Sink block in server mode, but don't connect to it, then destroy the block.

The socket created here is leaked:

https://github.com/gnuradio/gnuradio/blob/3349d483a1f7e5db0eea042815d237810373857b/gr-network/lib/tcp_sink_impl.cc#L180-L186

If a connection is accepted, then the socket is assigned to `d_tcpsocket` and eventually deleted. But if no connection is accepted, then nothing keeps track of `tmpSocket` and it is leaked.

To observe the problem, I used valgrind (`valgrind --leak-check=full --show-leak-kinds=definite python3`) and ran the following Python code:

```python
from gnuradio import network

t_sink = network.tcp_sink(1, 1, "localhost", 1234, 2)
t_sink.start()
t_sink.stop()
del t_sink
quit()
```

This produces the following report:

```
==2079297== 80 bytes in 1 blocks are definitely lost in loss record 1,760 of 2,425
==2079297==    at 0x4849013: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==2079297==    by 0x7BC502B: gr::network::tcp_sink_impl::connect(bool) (tcp_sink_impl.cc:171)
==2079297==    by 0x7BC21B0: gr::network::tcp_sink_impl::run_listener() (tcp_sink_impl.cc:116)
==2079297==    by 0x64EA2B2: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30)
==2079297==    by 0x4A43B42: start_thread (pthread_create.c:442)
==2079297==    by 0x4AD4BB3: clone (clone.S:100)
```

marcusmueller, 2024-02-14:
I think this was closed in #6697 ; running valgrind in the background, but `--leak-check=full` is not very fast.

marcusmueller, 2024-02-14:
Yeah, aside from the inherent Python leaks, valgrind now looks clean to me!

argilo, 2024-02-14:
#6697 doesn't look like it made any changes to the code in question. I'll do a quick test to see whether I still observe the leak.

argilo, 2024-02-14:
Yep, the memory leak is still there on `maint-3.10` and `main`.

ASAN's leak check also detects it:

```
Direct leak of 80 byte(s) in 1 object(s) allocated from:
    #0 0x7efcf66b61e7 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:99
    #1 0x7efcefd95503 in gr::network::tcp_sink_impl::connect(bool) /home/argilo/prefix_311/src/gnuradio/gr-network/lib/tcp_sink_impl.cc:171
```

argilo, 2024-02-14:
I see this is now tagged with "clarification needed". Are any more details needed?

marcusmueller, 2024-02-14:
ah, I added that tag so I don't forget to look at it when valgrind was finished, and then it didn't take as long as I thought.

Here's how I reproduced things:
https://gist.github.com/marcusmueller/966cb53e75a0dc73db3c89685b7e708c

Interestingly, when run non-interactively, `stop()` never finishes, as far as I can tell. It's really strange. And my log also doesn't contain any leaks in gr:: code!

marcusmueller, 2024-02-14:
Interactively, however, yes:

```
==74295== 88 bytes in 1 blocks are definitely lost in loss record 4,154 of 5,841
==74295==    at 0x4842F95: operator new(unsigned long) (vg_replace_malloc.c:483)
==74295==    by 0x6B300E4: gr::network::tcp_sink_impl::connect(bool) (gr-network/lib/tcp_sink_impl.cc:171)
==74295==    by 0x6B2FD90: gr::network::tcp_sink_impl::run_listener() (gr-network/lib/tcp_sink_impl.cc:116)
==74295==    by 0x71D11B2: execute_native_thread_routine (thread.cc:104)
==74295==    by 0x4E73946: start_thread (pthread_create.c:444)
==74295==    by 0x4EF9873: clone (clone.S:100)
```

[interactivelog.txt](https://github.com/gnuradio/gnuradio/files/14285494/interactivelog.txt)

marcusmueller, 2024-02-14:
To make this easier on future readers, this is the offending code:

```c++
    asio::ip::tcp::socket* tmpSocket = new asio::ip::tcp::socket(d_io_context);
    d_acceptor->async_accept(*tmpSocket,
                             [this, tmpSocket](const asio::error_code& error) {
                                 accept_handler(tmpSocket, error);
                             });
```

argilo, 2024-02-14:
> Interestingly, when run non-interactively, `stop()` never finishes, as far as I can tell

Yep, I think you've discovered another bug (deadlock) in the TCP Sink.

argilo, 2024-02-14:
GDB shows that `stop()` gets stuck here waiting for the listener thread to terminate:

https://github.com/gnuradio/gnuradio/blob/7d61746e27778c56eb8a805ad940c89ff265e313/gr-network/lib/tcp_sink_impl.cc#L206

argilo, 2024-02-14:
Looks like a race condition triggers this.

`start()` kicks off `run_listener()` in a thread, and that thread later sets `d_thread_running = true;` when it starts. But if `stop()` is called prior to the thread starting, then `d_thread_running` is still false and it doesn't set `d_stop_thread = true;`, so the listener thread continues forever.

argilo, 2024-02-14:
Perhaps `start()` should set `d_thread_running` instead of leaving it to the background thread to do that.

argilo, 2024-02-14:
I'll open another issue for this.

argilo, 2024-02-14:
#7133 is opened to track the race condition & deadlock.