gnuradio/gnuradio #6709: Does this support AirspyHF-Discovery?

joansola, 2023-06-07:
I am using this on Ubuntu 20.04, with gnuradio-companion.

I am a little lost since I am new to SDR, so I am unsure if this is the place for this question.

I have the HF-Discovery. I am using this on Ubuntu 20.04, with gnuradio-companion. I installed the following:

    github.com/libusb/libusb
    github.com/airspy/airspyhf
    github.com/pothosware/SoapyAirspyHF

In gnuradio-companion, I add the "Soapy AirspyHF Source" block, and connect it to a sink.

Then I edit the properties of the block until no errors remain (sample rate 768kHz).

On execution, I get this error on function `set_gain`:

[...]
my_file.py", line 96, in _init_
    self.soapy_airspyhf_source_0.set_gain(0, 'RF', min(max(0, -48.0), 0.0))
ValueError: source: Unknown gain RF for channel 0

I am unsure of where to look for causes of this. I hoped this comes from this lib, and hence the issue.

Versions:

gnuradio: 3.10.5.1
airspyhf: 1.7.1
Firmware: R3.0.6-CD

Note: I also posted this issue at `github.com/pothosware/SoapyAirspyHF`

joansola, 2023-06-07:
OK following on this, I thinkk I found a bug in `Settings.cpp` on SoapyAirspyHF:

#if AIRSPYHF_VER_MAJOR >= 1
#if AIRSPYHF_VER_MINOR >= 1
#if AIRSPYHF_VER_REVISION >= 2
#define HASGAINS
#endif
#endif
#endif

I actually have airspyhf v1.7.1, so I passx the first two IF, but not the last.

I replaced the third IF with value 1 instead of 2, and my issue with the RF gain disappeared.

I posted this bug there. Closing this.

closed 2023-06-07 by joansola

willcode, 2023-06-07:
@joansola I posted an issue and fix (PR) on SoapyAirspyHF recently.