gnuradio/gnuradio #6723: Python Gateway: work return code enums do not match return type of general_work

willcode, 2023-06-16:
### What happened?

The enums here https://github.com/gnuradio/gnuradio/blob/04d4f0d4a41bf8b4df86ef5feed3bc8ab0dd6118/gnuradio-runtime/python/gnuradio/gr/bindings/block_gateway.h#L32

Can not be returned here https://github.com/gnuradio/gnuradio/blob/04d4f0d4a41bf8b4df86ef5feed3bc8ab0dd6118/gnuradio-runtime/python/gnuradio/gr/bindings/block_gateway.h#L96

Error:
```
thread_body_wrapper :error: ERROR thread[thread-per-block[0]: <block EPB: File Source to Tagged Stream(3)>]: TypeError: '>' not supported between instances of 'gnuradio.gr.gr_python.work_return_t' and 'int'

At:
  /dcs/lib/python3.10/site-packages/gnuradio/gr/gateway.py(303): consume_items
  /dcs/lib/python3.10/site-packages/gnuradio/gr/gateway.py(200): handle_general_work
```

### System Information

Any

### GNU Radio Version

3.11-git (main)

### Steps to Reproduce the Problem

Return `gr.WORK_DONE` from a Python block.

willcode, 2023-06-16:
@mormj Any clue why returning -1 from an EPB does not terminate the flowgraph, or print the "we're done" message in block_executor.cc? Is this a PyBind11 inheritance problem? I can't see how it all fits together.

marcusmueller, 2024-02-14:
Can reproduce:

[image]
```Python
import numpy as np
from gnuradio import gr

class blk(gr.sync_block):  # other base classes are basic_block, decim_block, interp_block
    def __init__(self):  # only default arguments here
        gr.sync_block.__init__(
            self,
            name='instant return -1 block',   # will show up in GRC
            in_sig=[np.complex64],
            out_sig=[]
        )

    def work(self, input_items, output_items):
        return -1
```

happily keeps that window open. 
https://gist.github.com/marcusmueller/901bf36bbb0e52929a8db9a0b23d7e62

AsriFox, 2024-09-09:
Can also reproduce this behavior without GUI: if 'Run Options' is set to 'Prompt for Exit', the flowgraph will not terminate after the block returns -1. However, if 'Run to Completion' is selected, then it terminates as intended. Seems like the flowgraph only terminates after `tb.wait()`, and it's not called until 'Enter' is pressed or, in case of GUI, until the `closeEvent` happens.