gnuradio/gnuradio #6730: Crosscompilation of runtime/lib with target_precompile_headers enabled may fail

ardanew, 2023-06-22:
### What happened?

Compilation with some (buggy) cross-compilers fails when `https://github.com/gnuradio/gnuradio/blob/32b028bed849908699a9927bca54be9e6027e7ec/gnuradio-runtime/lib/CMakeLists.txt#LL155C12-L155C12` cmake tells that it can do precompiled headers, but compiler really can't.

Maybe it is better to add sync_block.h to `common-precompiled-headers` target, or provide some option to disable usage of target-precompile-header statement.

For now I set CMake version to old one to avoid this bug, but it is just a workaround.

### System Information

OS: Windows ---crosscompiling---> linux aarch64
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

It's really hard to reproduce, glitchy crosscompiler is required. Building with ENABLE_COMMON_PCH=OFF.

### Relevant log output

```shell
No errors in output, compiler just returns ERRORLEVEL=1
If I change -O3 to -O0 it compiles. 

`[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/constants.cc.o
cd C:/projects/gnuradio/build/aarch64_peta2_t9_cg5-release/gnuradio-runtime/lib && C:/Xilinx/Vitis/2021.1/gnu/aarch64/nt/aarch64-linux/bin/aarch64-linux-gnu-g++.exe --sysroot=c:/projects/sysroots/aarch64_peta2_t9_cg5 -DBOOST_DLL_USE_STD_FS -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DHAVE_COSF -DHAVE_GETPAGESIZE -DHAVE_MMAP -DHAVE_PTHREAD_SETSCHEDPARAM -DHAVE_PTHREAD_SIGMASK -DHAVE_SCHED_SETSCHEDULER -DHAVE_SHM_OPEN -DHAVE_SIGACTION -DHAVE_SIGNAL_H -DHAVE_SINCOS -DHAVE_SINCOSF -DHAVE_SINF -DHAVE_SYSCONF -DHAVE_SYS_IPC_H -DHAVE_SYS_MMAN_H -DHAVE_SYS_RESOURCE_H -DHAVE_SYS_SHM_H -DHAVE_SYS_TYPES_H -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DTRY_SHM_VMCIRCBUF -Dgnuradio_runtime_EXPORTS -IC:/projects/gnuradio/build/aarch64_peta2_t9_cg5-release/gnuradio-runtime/lib/math -IC:/projects/gnuradio/gnuradio-runtime/lib/../include -IC:/projects/gnuradio/build/aarch64_peta2_t9_cg5-release/gnuradio-runtime/lib/../include -IC:/projects/gnuradio/gnuradio-runtime/lib -IC:/projects/gnuradio/gnuradio-runtime/lib/pmt/../../include -IC:/projects/boost/output/aarch64_peta2_t9_cg5-release/include -IC:/projects/volk/output/aarch64_peta2_t9_cg5-release/include -IC:/projects/spdlog/output/aarch64_peta2_t9_cg5-release/include -IC:/projects/mpir/output/aarch64_peta2_t9_cg5-release/include -isystem c:/projects/zmq/output/aarch64_peta2_t9_cg5-release/include/zmq -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -O3 -DNDEBUG -std=c++17 -fPIC -Winvalid-pch -include C:/projects/gnuradio/build/aarch64_peta2_t9_cg5-release/gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/cmake_pch.hxx -MD -MT gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/constants.cc.o -MF CMakeFiles/gnuradio-runtime.dir/constants.cc.o.d -o CMakeFiles/gnuradio-runtime.dir/constants.cc.o -c C:/projects/gnuradio/build/aarch64_peta2_t9_cg5-release/gnuradio-runtime/lib/constants.cc

make.exe[2]: *** [gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/build.make:97: gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/constants.cc.o] Error 1`
```

marcusmueller, 2023-06-26:
Hm it seems the `-DENABLE_COMMON_PCH=Off` switch isn't working consistently :(

marcusmueller, 2024-03-11:
tagging low, we should just remove PCH