gnuradio/gnuradio #6732: i want increment Varible block, from python block

girs1982, 2023-06-25:
### Feature Description

how i can done it? how import and set values?

`

"""
Embedded Python Blocks:

Each this file is saved, GRC will instantiate the first class it finds to get          в секунде миллион микросекунд т.е делитель 2
ports and parameters of your block. The arguments to __init__  will be the
parameters. All of them are required to have default values!
"""
import time
import pmt
import numpy as np

from gnuradio import gr, grc

from datetime import datetime

class blk (gr.sync_block):
    """
    Block to decode the data on an already squared signal, comprised of 0's and 1's.
    """

    def __init__ (self
                , baseband_freq = 600e6
                , sample_rate = 2e6
                ,gsm900s = 789e6
                , sink_file = None):  # only default arguments here

        gr.sync_block.__init__(
            self,
            name = 'OOK to bin sink',
            in_sig = [np.complex64],
          #  in_sig = [],
            out_sig = []
        )
        self.portName = 'some out'
        self.message_port_register_out (pmt.intern (self.portName))

        self.threshold = (sample_rate / baseband_freq) / 2
        self.gsmfreq=gsm900s
    
       

                 

    def work (self, input_items, *args, **kwargs):    
       # print("work__________")  
       
        samples = input_items [0]        
        in0 = input_items[0]
        processed = 0    
        self.gsmfreq+= 1
        print(self.gsmfreq)
 #       gsmfreq+= 1
        print("DD")
   

        return len (samples)`

i want to uncrement freqncy to scan signals in auto mode

### Feature Urgency

high (impacts broad portion of ecosystem)

willcode, 2023-06-25:
Please ask for help on chat.gnuradio.org or the mailing list. The issue tracker is used for bug reports and feature requests. I do not see a feature request here, even though the Feature Request tag was added.

closed 2023-06-25 by willcode

girs1982, 2023-06-26:
> Please ask for help on chat.gnuradio.org or the mailing list. The issue tracker is used for bug reports and feature requests. I do not see a feature request here, even though the Feature Request tag was added.

I not know it, can yoou give link? i noobb in gnu radio

girs1982, 2023-06-26:
> Please ask for help on chat.gnuradio.org or the mailing list. The issue tracker is used for bug reports and feature requests. I do not see a feature request here, even though the Feature Request tag was added.

and i have banned in chat ,wtf?