gnuradio/gnuradio #6735: gr 3.10.6.0 fails to build with libfmt 10

hobbes1069, 2023-06-28:
### What happened?

Build fails on all arches

### System Information

Fedora Linux

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.6.0

### Steps to Reproduce the Problem

Attempt to build with libfmt 10.

### Relevant log output

```shell
/builddir/build/BUILD/gnuradio-3.10.6.0/gr-uhd/lib/rfnoc_rx_streamer_impl.cc:144:23:   required from here
/usr/include/fmt/core.h:1691:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1691 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1691:7: note: 'formattable' evaluates to false
```
https://kojipkgs.fedoraproject.org//work/tasks/8847/102698847/build.log
```

willcode, 2023-06-28:
@marcusmueller @mbr0wn could you check this out?
https://github.com/gnuradio/gnuradio/blob/70f71b5665ca05ce124bee3872fbbd995c4150b2/gr-uhd/lib/rfnoc_rx_streamer_impl.cc#L144

willcode, 2023-06-28:
@hobbes1069 If you comment out this one logging line, does the rest build with fmt v10? Is v10 going to be used in Fedora 40? It looks like it's currently v9 through Fedora 39.

hobbes1069, 2023-06-28:
The libfmt 10 build is ongoing as we speak. I'll try the fix when I get back home work. Thanks for the quick response!

mtasaka, 2023-06-28:
https://github.com/jasonabele/ettus-uhd/blob/3060006b35859c8766e00d3b234b51a19aa8595f/host/include/uhd/types/metadata.hpp#L74

`uhd::rx_metadata_t::error_code_t` is just a enum so the simplest fix is perhaps just to cast this to int or so like
```
        d_logger->warn("RFNoC Streamer block received error {:s} (Code: {})",
                       d_metadata.strerror(),
                       static_cast<int>(d_metadata.error_code));
```

mtasaka, 2023-06-28:
Actually the above change fixes build with fmt10: https://koji.fedoraproject.org/koji/taskinfo?taskID=102709196

willcode, 2023-06-28:
@mtasaka Good! Are you planning to put in a PR? Otherwise, I can.

mtasaka, 2023-06-29:
Okay, now I've submitted PR.

PR #6736 (2023-06-29): gr-uhd/lib: support fmtlib 10
In rfnoc_rx_streamer_impl::work , d_metadata.error_code has type uhd::rx_metadata_t::error_code_t , which is just enum.

But with fmtlib 10, fmt::format won't accept enum by default. So just cast this variable error_code to int to make fmtlib-10 accept this.

Closes #6735 .

- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/blob/main/CONT [... cut]
commit b6c147dc (2023-06-29): gr-uhd/lib: support fmtlib 10

In rfnoc_rx_streamer_impl::work , d_metadata.error_code has
type uhd::rx_metadata_t::error_code_t , which is just enum.

But with fmtlib 10, fmt::format won't accept enum by default.
So just cast this variable error_code to int to make fmtlib-10
accept this.

Closes #6735 .
commit a231e7c8 (2023-06-29): gr-uhd/lib: support fmtlib 10

In rfnoc_rx_streamer_impl::work , d_metadata.error_code has
type uhd::rx_metadata_t::error_code_t , which is just enum.

But with fmtlib 10, fmt::format won't accept enum by default.
So just cast this variable error_code to int to make fmtlib-10
accept this.

Closes #6735 .

(cherry picked from commit b6c147dc1b78b706a2ebb6d78e5199f3ad0f8e83)
closed 2023-06-29 by willcode