gnuradio/gnuradio #6755: gnuradio-3.10.7.0 breaks gnuradio-companion gain control with gr-soapy

ChrisVine, 2023-07-16:
### What happened?

The use of a Qt GUI range widget to control manually the gain field of the SoapySDRPlay source in gnuradio-companion no longer works with gnuradio-3.10.7.0 and gr-soapy using the following components:

sdrplay 2.13.1 proprietary driver
SoapySDR as at commit of 2022.12.25
SoapySDRPlay2 as at commit of 2021.11.21
gnuradio-3.10.7.0
volk-3.0.0
RSP1A SDR device

The gain setting made in gnuradio-companion is ignored and has no effect.  The flowgraph operates with an immutable and low gain setting which has no regard to either the range widget's starting setting nor any changes made while the flowgraph is running.  Other variable settings still appear to work normally.  The AGC and Bias tee fields are set to False.

By reverting to gnuradio-3.10.6.0 it works fine, so this may be connected with the release note indicating that there is "Better AGC and gain behavior in RTL, AirspyHF and SDRPlay blocks".  This may have broken something in gr-soapy.

### System Information

OS: Linux (6.4.3 kernel)
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

Start gnuradio-companion and run a flowgraph with the components mentioned above.

### Relevant log output

```shell
No logging information relevant to the issue is provided.
```

willcode, 2023-07-16:
Hi @ChrisVine - I changed the gains to be positive, to be more like the other blocks. The gain now has a range of 0-39dB. Let me know if that fixes it.

ChrisVine, 2023-07-16:
Yes, changing to a positive range fixes the problem.  I leave it to you when you want to close this issue: there may be other lost souls having problems with the gr-soapy block with SoapySDRPlay.

willcode, 2023-07-16:
I knew there would be a few people using it, but since it was so broken people were going to have to adapt to the separate LNA setting and AGC setpoint anyway. The Issue will still show when closed, so I'll close it now. Glad that worked for you, and send feedback. The Soapy blocks were a quick hack when we first put in gr-soapy, and I just got around to starting to update them.

closed 2023-07-16 by willcode