gnuradio/gnuradio #6766: Crash in QT GUI frequency sink in 3.10.5.1

sq5bpf, 2023-07-30:
### What happened?

What happened:
A flowgraph using the QT GUI frequency sink crashes when repeatedly changing "Center frequency" via the xmlrpc server. 
Narrowed it down to a very simple test case with just a noise source fed through a throttle into a QT GUI frequency sink (see test3.grc), and a script to hammer the xmlrpc interface (see tt.py).

The original flowgraph i have problems with  has more QT widgets. I'm also enclosing it for reference. Before crashing it also has an error where the QT widgets don't update correctly (for example the frequency is changed via xmlrpc, and the receiver is indeed retuned, but the widgets don't see a change of frequency). This usually happens when the values are changed by the telive program, which does multiple of requests very fast one after the other.

What i expect to happen:
I expect that it works without a crash. Also expect all QT elements to update correctly. 

Other:
The only reference to a similar problem i've found is this:
https://www.mail-archive.com/discuss-gnuradio@gnu.org/msg70266.html  (this had no followup).

Enabling "Thread-safe setters" in the original flowgraph didn't help.

I would also be grateful if someone posts a workaround that could be used so that this can be used with an unfixed gnuradio version.

### System Information

OS: Debian 12 x86_64 (latest)
GR Installation Method: distro deb package

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.5.1

### Steps to Reproduce the Problem

Install Debian 12, and gnuradio from the distribution packages.
Run test3.grc
Stretch the window so that it's about 3 times wider (no idea why this is needed)
Run tt.py
Wait for the crash. It usually occurs when the counter displayed by tt.py is < 400.
If it doesn't crash, then play around with the window size. 

In case this is a problem with QT theming: i'm running this under xfce.

[problem.tar.gz](https://github.com/gnuradio/gnuradio/files/12208868/problem.tar.gz)

### Relevant log output

```shell
Lots of:
127.0.0.1 - - [29/Jul/2023 23:49:10] "POST /RPC2 HTTP/1.1" 200 -
QCoreApplication::sendPostedEvents: Cannot send posted events for objects in another thread

And then some error message (the error message at the end can be different or none at all). Some error messages i've seen from this example flowgraph, or another more complicated flowgraph:

double free or corruption (fasttop)
malloc(): unaligned tcache chunk detected

And then a segfault:

>>> Done (return code -11)
```

marcusmueller, 2023-07-30:
Good news is that the linked mail thread is almost certainly unrelated :)
Anyway, reproducing this on Fedora 37, main branch. Interesting.

marcusmueller, 2023-07-30:
```
$ gdb --args python ./test3.py
…
(gdb) run
…
// run tt.py
malloc_consolidate(): unaligned fastbin chunk detected

Thread 8 "python" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff92ffd6c0 (LWP 83041)]
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
Downloading source file /usr/src/debug/glibc-2.37-4.fc38.x86_64/nptl/pthread_kill.c
44	      return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff78b08b3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff785fabe in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff784887f in __GI_abort () at abort.c:79
#4  0x00007ffff784960f in __libc_message (fmt=fmt@entry=0x7ffff79c352f "%s\n") at ../sysdeps/posix/libc_fatal.c:150
#5  0x00007ffff78ba775 in malloc_printerr (str=str@entry=0x7ffff79c6130 "malloc_consolidate(): unaligned fastbin chunk detected") at malloc.c:5651
#6  0x00007ffff78bb30c in malloc_consolidate (av=av@entry=0x7fff84000030) at malloc.c:4736
#7  0x00007ffff78bd958 in _int_malloc (av=av@entry=0x7fff84000030, bytes=bytes@entry=8192) at malloc.c:3951
#8  0x00007ffff78bec34 in __GI___libc_malloc (bytes=8192) at malloc.c:3305
#9  0x00007ffff7ba513c in _PyMem_RawMalloc (size=<optimized out>, ctx=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/obmalloc.c:101
#10 PyMem_RawMalloc (size=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/obmalloc.c:586
#11 _PyObject_Malloc (ctx=<optimized out>, nbytes=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/obmalloc.c:2003
#12 _PyObject_Malloc (nbytes=<optimized out>, ctx=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/obmalloc.c:1996
#13 PyMem_Malloc (size=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/obmalloc.c:623
#14 0x00007ffff7c58f65 in _buffered_init (self=self@entry=0x7fffa4373d70) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Modules/_io/bufferedio.c:718
#15 0x00007ffff7c58e18 in _io_BufferedReader___init___impl (buffer_size=8192, raw=0x7fffa3eecf10, self=0x7fffa4373d70) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Modules/_io/bufferedio.c:1431
#16 _io_BufferedReader___init__ (self=0x7fffa4373d70, args=<optimized out>, kwargs=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Modules/_io/clinic/bufferedio.c.h:435
#17 0x00007ffff7bb0b4b in type_call (type=<optimized out>, args=0x7fffa3ec4340, kwds=0x0) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/typeobject.c:1112
#18 0x00007ffff7bb0953 in _PyObject_MakeTpCall (tstate=0x555556c0dad0, callable=0x7ffff7ed7440 <PyBufferedReader_Type>, args=<optimized out>, nargs=2, keywords=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/call.c:214
#19 0x00007ffff7bb92a8 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Python/ceval.c:4774
#20 0x00007ffff7bb536a in _PyEval_EvalFrame (throwflag=0, frame=0x7fffe986e3b0, tstate=0x555556c0dad0) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Include/internal/pycore_ceval.h:73
#21 _PyEval_Vector (tstate=0x555556c0dad0, func=func@entry=0x7fffa3e5fe20, locals=locals@entry=0x0, args=<optimized out>, argcount=argcount@entry=4, kwnames=kwnames@entry=0x0) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Python/ceval.c:6439
#22 0x00007ffff7bb36ef in _PyFunction_Vectorcall (kwnames=0x0, nargsf=<optimized out>, stack=<optimized out>, func=0x7fffa3e5fe20) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/call.c:393
#23 _PyObject_FastCallDictTstate (tstate=0x555556c0dad0, callable=0x7fffa3e5fe20, args=<optimized out>, nargsf=<optimized out>, kwargs=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/call.c:141
#24 0x00007ffff7bdc573 in _PyObject_Call_Prepend (kwargs=0x0, args=0x7fffa3ef0800, obj=0x7fffa548fed0, callable=0x7fffa3e5fe20, tstate=0x555556c0dad0) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/call.c:482
#25 slot_tp_init (self=0x7fffa548fed0, args=0x7fffa3ef0800, kwds=0x0) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/typeobject.c:7863
#26 0x00007ffff7bb0b4b in type_call (type=<optimized out>, args=0x7fffa3ef0800, kwds=0x0) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/typeobject.c:1112
#27 0x00007ffff7bb0953 in _PyObject_MakeTpCall (tstate=0x555556c0dad0, callable=0x5555569b5f00, args=<optimized out>, nargs=3, keywords=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/call.c:214
#28 0x00007ffff7bb92a8 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Python/ceval.c:4774
#29 0x00007ffff7bb536a in _PyEval_EvalFrame (throwflag=0, frame=0x7fffe986e188, tstate=0x555556c0dad0) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Include/internal/pycore_ceval.h:73
#30 _PyEval_Vector (tstate=0x555556c0dad0, func=func@entry=0x7fffa3e5e020, locals=locals@entry=0x0, args=args@entry=0x7fff92ffcbf8, argcount=argcount@entry=1, kwnames=kwnames@entry=0x0) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Python/ceval.c:6439
[... cut ...]
#39 method_vectorcall (method=<optimized out>, args=0x7ffff7ef3870 <_PyRuntime+58928>, nargsf=<optimized out>, kwnames=0x0) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Objects/classobject.c:67
#40 0x00007ffff7c97d10 in thread_run (boot_raw=0x7fffa3eecc30) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Modules/_threadmodule.c:1092
#41 0x00007ffff7c65c58 in pythread_wrapper (arg=<optimized out>) at /usr/src/debug/python3.11-3.11.4-1.fc38.x86_64/Python/thread_pthread.h:241
#42 0x00007ffff78ae907 in start_thread (arg=<optimized out>) at pthread_create.c:444
#43 0x00007ffff7934870 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

marcusmueller, 2023-07-30:
The other threads are all blocked in futexes (the syscall is also wait-related): 
```
(gdb) info threads
  Id   Target Id         Frame
* 1    LWP 83041         __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
  2    LWP 80547         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x7fffffffc880, op=137, expected=0, futex_word=0x7ffff7ee53f0 <_PyRuntime+432>) at futex-internal.c:57
  3    LWP 82798         0x00007ffff792735d in __GI___poll (fds=0x7fff9c001660, nfds=3, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  4    LWP 82799         0x00007ffff792735d in __GI___poll (fds=fds@entry=0x7fffa305fd28, nfds=nfds@entry=1, timeout=timeout@entry=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  5    LWP 83034         syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
  6    LWP 83035         0x00007ffff792735d in __GI___poll (fds=0x555556b58a40, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  7    LWP 83036         0x00007ffff792735d in __GI___poll (fds=0x7fff88000b90, nfds=3, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  8    LWP 83040         0x00007ffff792735d in __GI___poll (fds=0x7fff80000b90, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  9    LWP 83045         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555556faa814) at futex-internal.c:57
  10   LWP 83046         0x00007ffff78f8413 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7fff91df6770, rem=0x7fff91df6770) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:48
  11   LWP 83047         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x7fff915f5b08, op=137, expected=0, futex_word=0x555556ec2b64) at futex-internal.c:57
  12   LWP 83459         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555556e63800) at futex-internal.c:57
  13   LWP 83460         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555556f7bfb8) at futex-internal.c:57
  14   LWP 83461         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555556f7bfb8) at futex-internal.c:57
  15   LWP 83462         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555556f7bfb8) at futex-internal.c:57
  16   LWP 83463         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555556f7bfb8) at futex-internal.c:57
  17   LWP 83464         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555556f7bfb8) at futex-internal.c:57
  18   LWP 83465         0x00007ffff78ab1d9 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x555556f7bfb8) at futex-internal.c:57
```

iwbnwif, 2023-09-23:
I am having a very similar problem setting the center frequency of a QT GUI Frequency Sink from a ZMQ message source via a Message Pair to Var block. Output error `QCoreApplication::sendPostedEvents: Cannot send posted events for objects in another thread` followed by a crash (return code -11) after a seemingly random number of messages.

Please post a work around if one has been found.

marcusmueller, 2023-09-23:
@iwbnwif no workaround; this requires rearchitecting of gr-qtgui :(

marcusmueller, 2023-09-23:
@iwbnwif but a start would be not to use Message pair to var; instead, at least send messages directly to the qt gui frequency sink.

sq5bpf, 2023-10-02:
Seems this bug will not get fixed anytime soon. 

So I'd like to ask again if there is some workaround that i could do in the flowgraph, such as adding some python code to the flowgraph to do some safe setting of this variable and maybe some additional QT magic? This is very basic functionality, so i guess there must be some undocumented common-knowledge how to use xml-rpc with qt-gui without crashes.

I really need this functionality to provide at least a partial replacement for the wx-gui version of the flowgraphs used in telive. I guess you have already heard various user comments about the comparison of wx-gui to qt-gui functionality, so i will spare mine.

marcusmueller, 2023-10-02:
@sq5bpf  it's a bit too late around these parts of the globe, but just so you know you've not been forgotten. We're currently going through all issues and assessing their severity and how urgently they need to be fixed, and what takeaways for newer architectures are. Your issue will rank on that list, pretty certain.

iwbnwif, 2024-01-21:
> @iwbnwif but a start would be not to use Message pair to var; instead, at least send messages directly to the qt gui frequency sink.

Thank you, I will look at taking this approach.

BogdanDIA, 2024-05-16:
Ah, landed here as I got the same problem in gr 3.10.7.1

sq5bpf, 2024-05-26:
> @iwbnwif but a start would be not to use Message pair to var; instead, at least send messages directly to the qt gui frequency sink.

Could you post a simple example flowgraph where a variable is set via xmlrpc (and not message passed via zmq), and this generates a message to the qt gui frequency sink to change the frequency? I'm sure this would help many people.

And btw is there any progress in fixing this bug?

sq5bpf, 2024-05-27:
Found the workaround using variable to message block. Big thanks to @marcusmueller for the comment about using a message. The below flowgraph will not crash when hammered for a few minutes via xmlrpc with the tt.py script published above (tested on gnuradio 3.10.5.1)
[test3_fixed.grc.gz](https://github.com/gnuradio/gnuradio/files/15454837/test3_fixed.grc.gz)

This is enough for my usecase, hope it will help others having the same problem.