gnuradio/gnuradio #6774: Build error related to libfmt

krzygorz, 2023-08-05:
### What happened?

The main branch of GNU radio fails to build with gcc 13.1.1, libfmt 10.0.0 and spdlog 1.11.0. This seems to be similar to #6278. The error:
```
/tmp/gnuradio/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc:258:24:   required from here
/usr/include/fmt/core.h:1691:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1691 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1691:7: note: ‘formattable’ evaluates to false
```

Commenting out this line makes the compilation succeed:
```
    debug_logger->error("ERROR Don't know how to handle Knob Type (from): {}", knob.type);
```

### System Information

OS: Arch Linux
GR Installation Method: from source (commit 35f8a0b, main branch)
gcc 13.1.1
libfmt 10.0.0
spdlog 1.11.0

### GNU Radio Version

3.11-git (main)

### Specific Version

g35f8a0b

### Steps to Reproduce the Problem

```
git clone https://github.com/gnuradio/gnuradio/
mkdir build
cmake -B build -S gnuradio CMAKE_BUILD_TYPE=Debug
cd build
make
```

### Relevant log output

```shell
[  0%] Built target common-precompiled-headers
[  1%] Built target gnuradio-pmt
[  1%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/tpb_thread_body.cc.o
[  1%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/transfer_type.cc.o
[  1%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/vmcircbuf.cc.o
[  1%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/vmcircbuf_createfilemapping.cc.o
[  1%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_shm_open.cc.o
[  1%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/vmcircbuf_mmap_tmpfile.cc.o
[  1%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/vmcircbuf_prefs.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/vmcircbuf_sysv_shm.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/messages/msg_accepter_msgq.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/messages/msg_producer.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/messages/msg_queue.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/thread/thread.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/thread/thread_body_wrapper.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/thread/thread_group.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/math/fast_atan2f.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/math/fxpt.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/math/random.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/rpcmanager.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_aggregator.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_booter_aggregator.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_selector.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcserver_thrift.cc.o
[  2%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcpmtconverters_thrift.cc.o
In file included from /usr/include/spdlog/fmt/fmt.h:31,
                 from /tmp/gnuradio/gnuradio-runtime/lib/../include/gnuradio/io_signature.h:16,
                 from /tmp/gnuradio/gnuradio-runtime/lib/../include/gnuradio/basic_block.h:15,
                 from /tmp/gnuradio/gnuradio-runtime/lib/../include/gnuradio/block.h:17,
                 from /tmp/gnuradio/gnuradio-runtime/lib/../include/gnuradio/sync_block.h:15,
                 from /tmp/build/gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/cmake_pch.hxx:5,
                 from <command-line>:
/usr/include/fmt/core.h: In instantiation of ‘constexpr fmt::v10::detail::value<Context> fmt::v10::detail::make_value(T&&) [with Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; T = const GNURadio::BaseTypes::type&]’:
/usr/include/fmt/core.h:1711:29:   required from ‘constexpr fmt::v10::detail::value<Context> fmt::v10::detail::make_arg(T&&) [with bool IS_PACKED = true; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; type <anonymous> = fmt::v10::detail::type::custom_type; T = const GNURadio::BaseTypes::type&; typename std::enable_if<IS_PACKED, int>::type <anonymous> = 0]’
/usr/include/fmt/core.h:1827:77:   required from ‘constexpr fmt::v10::format_arg_store<Context, Args>::format_arg_store(T&& ...) [with T = {const GNURadio::BaseTypes::type&}; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; Args = {GNURadio::BaseTypes::type}]’
/usr/include/fmt/core.h:1844:31:   required from ‘constexpr fmt::v10::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<T>::type>::type ...> fmt::v10::make_format_args(T&& ...) [with Context = basic_format_context<appender, char>; T = {const GNURadio::BaseTypes::type&}]’
/usr/include/spdlog/logger.h:374:75:   required from ‘void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const GNURadio::BaseTypes::type&}; spdlog::string_view_t = fmt::v10::basic_string_view<char>]’
/usr/include/spdlog/logger.h:90:13:   required from ‘void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const GNURadio::BaseTypes::type&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const GNURadio::BaseTypes::type&>]’
/usr/include/spdlog/logger.h:96:12:   required from ‘void spdlog::logger::log(spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const GNURadio::BaseTypes::type&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const GNURadio::BaseTypes::type&>]’
/usr/include/spdlog/logger.h:170:12:   required from ‘void spdlog::logger::error(fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const GNURadio::BaseTypes::type&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const GNURadio::BaseTypes::type&>]’
/tmp/gnuradio/gnuradio-runtime/lib/../include/gnuradio/logger.h:209:24:   required from ‘void gr::logger::error(fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const GNURadio::BaseTypes::type&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const GNURadio::BaseTypes::type&>]’
/tmp/gnuradio/gnuradio-runtime/lib/controlport/thrift/rpcpmtconverters_thrift.cc:258:24:   required from here
/usr/include/fmt/core.h:1691:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1691 |       formattable,
      |       ^~~~~~~~~~~
/usr/include/fmt/core.h:1691:7: note: ‘formattable’ evaluates to false
make[2]: *** [gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/build.make:1147: gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/thrift/rpcpmtconverters_thrift.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3839: gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

ryanvolz, 2023-08-14:
Conda-forge is in the process of migrating to libfmt 10, so I'm now seeing this in the attempted rebuild. Build logs [here](https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=759733&view=logs&jobId=d0d954b5-f111-5dc4-4d76-03b6c9d0cf7e&j=d0d954b5-f111-5dc4-4d76-03b6c9d0cf7e&t=6d4b912b-175d-51da-0fd9-4d30fe1eb4e7) if it helps.

willcode, 2023-08-15:
We should probably do the same as in #6288. I don't think the integer type makes for a useful error message.

PR #6808 (2023-08-28): Fix build with fmt-10.1
## Description
fmt doesn't format enums implicitly any more, causing the build to fail. Calling `to_string` fixes it.

## Related Issue
Fixes #6774

## Checklist

- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [ ] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md#dco-signed)
- [ ] My code follows the code style of this project. See [GREP1.md](https://github.com/gnuradi [... cut]
closed 2023-08-28 by willcode