gnuradio/gnuradio #6777: Quick Drop Option for GNU RADIO

GentlemanSch, 2023-08-07:
### Feature Description

Hello all,

I'm coming from an Mechanical engineering space and am just getting used to GNURADIO, going through the tutorials, I keep going through the loop of:

* Hit search
* Click icon
* Move Icon into place

Previously, I used a software called "LabVIEW," that was also graphical in nature, but had the ability to quickly add and link segments via:

* ctrl+[space]
* type description
* enter/return to select "node," (node sticks to mouse)
* R-click to drop node in place

Does something like that exist in GNU Radio now (if so I couldn't find it, but I'm new) 
If not, I think it would be a huge productivity boost to people using the software.

[LabVIEW Quickdrop Tutorial](https://www.ni.com/en/support/documentation/supplemental/08/boost-labview-productivity-with-quick-drop.html)

Thank you for your time!

### Feature Urgency

medium (would be nice to have in the near future)

willcode, 2023-08-07:
I can't see why adding keystrokes to the process makes it more efficient than dragging a block from the block list. Could you describe what part of this procedure is easier for you, and not just different from LabVIEW?

CTRL-F is the same as the search button, if that's the difference.

GentlemanSch, 2023-08-08:
Ok, CTRL + F helps but there is still some functionality Quick drop has that could be useful.

I admit, in the LabVIEW space, I got to tell people I trained, "you'll see, just give it a go." Mostly It's less about adding keystrokes than removing mouse movements. There are also plenty of articles from the LabVIEW space that talk about how useful it is as well (e.g. https://digilent.com/blog/why-you-need-to-use-quick-drop-if-youre-using-labview/ ), however I'll do my best to articulate why it is so useful beyond "trust me bro."

1) Searching for a "node" 
  - GNU Radio: CTRL+F, -> type in side window
    - User is looking at the right side of their screen to find the correct node rather than at where they want to put it. 
  - LV QuickDrop: ctrl+[space] -> window appears at cursor
    - User maintains focus on the space where they intend to place the node, aiding "flow," significantly

2) Selecting a "node:" (example "Throttle,")
  - GNU Radio: Type "Thr," look through the list, select the correct item.
    - Since "Throttle (old)," will always superspeed "Throttle," the user will not be able to use the enter key to populate the node on the map
   - LV QuickDrop (Equivalent): Type "Thr," -> down arrow -> enter key
     - Even though there are multiple items with a similar name, users can navigate through them via keyboard "up," and "down," keys.

3) Placing a "node"
  - GNU Radio: Enter key, double click, place the node in the middle of the diagram. Click + Drag from search is required for accurate placement.
  - LV QuickDrop (best recollection): Enter -> populates the node at the mouse location. Single Click -> "attaches," the node to the cursor. Node is placed with a subsequent single click. Node is eliminated without placing via "escape,"

willcode, 2023-08-09:
Other programs (e.g., Blender) also use spacebar-initiated placement. I don't see the speed at which people can drag/drop blocks as being a major part of constructing a GNU Radio flowgraph, but let's leave this open to collect other opinions.