gnuradio/gnuradio #6780: RFNoC mode tool at gnuradio v3.10

KETI-A, 2023-08-09:
### What happened?

I could not use gr-ettus's rfnocmodtool at GNU Radio v3.10 because could not build with v3.10.
I found that swig has been removed since v3.9.

How could I make rfnoc block for gnuradio v3.10?

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

1. build and install gnuradio v3.10.7
2. build is failed of gr-ettus

willcode, 2023-08-09:
Please ask for help on the usrp-users mailing list. gr-ettus is not a part of GNU Radio.

closed 2023-08-09 by willcode

KETI-A, 2023-08-09:
Thank you for your answer. I would like to ask about it on the usrp-users mailing list.