gnuradio/gnuradio #6783: QT GUI Time Raster Sink - missing MSG port

i3vfj, 2023-08-10:
### Feature Description

Hi everybody,,I'm I am working on a larger project, a "remote" transceiver for satellite traffic, trying to limit the need for bandwidth via Internet since the remote location does not have enough bandwidth available.
To further reduce bandwidth consumption and to have a spectral representation I do not send the IQ stream to the control terminal, but only the vectors obtained from the FFT; so I went from about 20Mb/s to 500kb/s, I would say more than acceptable!
In the control terminal, however, I have the need to "return" the "point and click" frequency indication to the remote side. This works correctly on the QT GUI Vector Sink ( spectral representation ), which has an MSG port but, from an operational point of view, it would be better to be able to have the same functionality via the QT GUI Time Raster Sink ( waterfall representation ) block, which does not have an MSG port available, even if the Xval value it's there.

I realize that this is not a "real problem" but just a missing feature that I hope can be implemented in the future ( I'm not so skilled to do it myself ;-) )
In the meantime, I continue to have fun with this project!

Thank you for your attention, maybe someone can suggest me some other solution.
 de Vittorio, I3VFJ

### Feature Urgency

medium (would be nice to have in the near future)

willcode, 2023-08-14:
Could you rewrite this to put the feature request at the top of the text so it's easier to find?

i3vfj, 2023-08-14:
Done, I think... I'm not an expert on GITHUB :-)
Please, let me know if now it's ok.
Tnx de Vittorio, I3VFJ

fingerprint: fb5f492a54e016c632c933d3ee4b7e38203c79ca

Inviato con l'email sicura [Proton Mail](https://proton.me/).

i3vfj, 2023-08-14:
Done, I think... I'm not an expert on GITHUB :-)
Please, let me know if now it's ok.
Tnx de Vittorio, I3VFJ