gnuradio/gnuradio #6792: #6616 breaks using selector blocks for bypass

daniestevez, 2023-08-17:
### What happened?

I typically use selector blocks to bypass/not-bypass at runtime a block (or chain of blocks) as follows:
```
              | ---------> some blocks -----> |
---> selector |                               | selector ---->
              | ----------------------------> |
```
The two selectors use the same variable to control the output index / input index, and by toggling the variable we bypass "some blocks" at runtime or not.

I've found that this usage is broken in 3.10.7.0, apparently because of #6616. I've built e33e70921285584bb44a24623c0fed5b7307ebed (the commit preceeding) #6616 in the `maint-3.10` branch, and this works fine.

In 3.10.7.0, when the variable is toggled at runtime, the flowgraph just hangs.

Pinging @marcusmueller, since this was his PR.

### System Information

OS: Arch linux
GR Installation Method: Package manager for 3.10.7.0, built from source for custom commit

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

A flowgraph is included to show this problem. This runs fine pre-#6616, but with #6616 it hangs (there is no data flow) as soon as the variable is toggled.

[test_bypass.zip](https://github.com/gnuradio/gnuradio/files/12368613/test_bypass.zip)

willcode, 2023-08-17:
As a guess, the switches do not happen at exactly the same time causing a freeze. Then the second switch never gets called. I don't see any obvious errors in the code.

@daniestevez As a debugging step, could you try removing the `min()` in `general_work()` so that `ninput_items[in_index]` is always consumed?

daniestevez, 2023-08-18:
> As a guess, the switches do not happen at exactly the same time causing a freeze.

This is bound to happen, but I don't see why it could/should cause a freeze. If the first switch toggles first, then for some time no one will consume its output and it will block incoming samples. Meanwhile, the second switch will starve. Eventually the second switch will toggle and start consuming samples, so data will flow again. If the second switch toggles first, then it might become starved for some time, producing no output. Meanwhile, the first switch will block because its output is not being consumed. At some point the first switch will toggle and produce output for the second switch, so data will flow again.

daniestevez, 2023-08-18:
> As a debugging step, could you try removing the min() in general_work() so that ninput_items[in_index] is always consumed?

Just tested this and the flowgraph still gets blocked.

daniestevez, 2023-08-18:
I've [added some printf debugging](https://github.com/daniestevez/gnuradio/commit/d1cbed99da8f7a1ee6d6e748755cc3885cd03de9) and a dirty hack that makes this work.

Apparently the deadlock happens because when the toggle happens, the `forecast()` of the second selector is called, asking how would it take to produce `N` samples (say `N = 8192`). The `forecast()` replies that it needs `N` elements in the second input. The scheduler continues calling `forecast()` with decreasing values of `N` until reaching `N = 1`, and seeing that it doesn't have the input samples required by the second selector at the moment, it keeps looping forever. The `forecast()` of the first selector is never called, so the samples that are needed are never produced.

It seems to me that the problem here is that the second selector could actually get some samples on its first input (there are samples produced immediately before the toggle). That seems to be the reason why the scheduler doesn't call the first selector. Probably the first selector output buffer is already full because samples have become stuck in the path involving the Multiply Const block.

The dirty hack that I did to work around this situation is, when `forecast()` is asked to produce exactly 1 output item, we lie and say that we can produce it for free (without requiring any input items at all). This causes the second selector `general_work()` to be called. In there, I also treat `noutput_items == 1` as a special case, consuming all the available samples in all the inputs in this case (still no output samples get produced if there are no available samples in the active input). This achieves draining the Multiply Const block and first selector output buffers, so after a couple calls of this special case, first selector gets called and produces samples into the input we need.

daniestevez, 2023-08-18:
Looking more carefully at the code, I see that the `gr::thread::scoped_lock l(d_mutex);` in `general_work()` should be before `to_copy` is calculated (i.e., before the first time we access `d_input_index`). Also, it would be good to have a `scoped_lock` in `forecast()` as well. However, these changes do not solve the problem.

daniestevez, 2023-08-18:
I think the key of why this worked before, with the broken selector (now that I've read carefully the PR, I agree that the selector was broken before), is what happens (typically, I guess) during the toggle:
```
[selector] forecast(8184) d_input_index = 0, d_output_index = 0 = {8184, }
[selector] general_work() d_input_index = 0, d_output_index = 0 finished: noutput_items = 8184, to_copy = 8184
        ninput_items[] = {16383, }
[selector] forecast(8192) d_input_index = 1, d_output_index = 0 = {0, 8192, }
[selector] forecast(4096) d_input_index = 1, d_output_index = 0 = {0, 4096, }
[selector] forecast(2048) d_input_index = 1, d_output_index = 0 = {0, 2048, }
[selector] forecast(1024) d_input_index = 1, d_output_index = 0 = {0, 1024, }
```

The `general_work()` is called with a `noutput_items` smaller than `ninput_items[0]`. The previous broken implementation consumed all the input, so there was no input left at the moment that the toggle happens. After the PR, there is some input left, which is what makes the flowgraph become stuck.

daniestevez, 2023-08-18:
In order to propose a solution that is not hacky, I'm trying to understand/define what should the Selector block actually do. Conceptually it is just like an electrical switch, but things get tricky when we speak in terms of GNU Radio buffers and such.

For instance, with the 3.10.7 Selector (and also with my workaround), this flowgraph gets stuck quickly. There are no switches at runtime here. The problem is that the Selector consumes on inactive inputs at the same rate as the active input, and here the inactive input arrives at twice the rate as the active input. This raises the question: should the selector consume all available input (i.e., provide no backpressure) on inactive inputs as it did before, or are there cases where some degree of backpressure is necessary? (I don't of a real world example where something like the flowgraph below would show up. I just crafted it to show this corner case).

[image]

daniestevez, 2023-08-18:
After thinking some more, my proposed solution is:

* `general_work()` should consume all available samples from inactive inputs, and only the samples that were copied from the active input.
* `forecast()` should lie and say that 1 output item can be produced for free, to ensure that `general_work()` gets called to consume all inactive inputs and prevent getting stuck in cases such as the one that motivated this issue.

I don't like the `forecast()` hack too much, but I don't know of a better way to ensure that `general_work()` gets called, because `forecast()` can't say something like: "I'd like `general_work()` to be called either if there are `noutput_items` in `d_input_index` or either if there are some samples on any of the other inputs.

willcode, 2023-08-18:
Thanks for working on this. I agree that forecast hack may be too hacky (will cause spurious calls to general_work), but don't have a better idea yet.

There are use cases where using up all input isn't right either (switching between file sources comes to mind).

More thought needed. And mostly AFK until next week.

daniestevez, 2023-08-18:
> When you have [toggle between SDR RX and signal source]

What would be the expected behaviour in this case? If I understand correctly, the selector doesn't provide backpressure to the signal source whenever it is the active input. Why should it do so when it is the inactive input?

> has a formally visible internal state machine per input that cleanly handles things like state and behaviour changes through tags and messages

Can you elaborate on this?

In a way your conclusion sounds reasonable, but in another way it makes me feel that we're asking this new hypothetical selector to do too many different things at the same time. Following that line of reasoning, we could also try to have the option to set relative rates of consumption on the inputs (https://github.com/gnuradio/gnuradio/pull/6616#issuecomment-1496969070).

My take is that, for simplicity, the selector should do the following regarding backpressure:
- For the active input, apply the backpressure coming from the output.
- For inactive inputs, do not apply any backpressure (which means all items are consumed as fast as possible)

Trying to apply additional backpressure relations between the inputs is perhaps best achieved using other blocks, external to the selector. These blocks can be useful in other situations, and can perhaps be more flexible in how they can be combined. For example, to make two signals always flow at exactly the same rate, you would use a "synchronize" block, which doesn't currently exist in-tree, but can be just a sync block with 2 inputs and 1 output that copies the first input to the output (and it can also be crudely achieved with in-tree blocks by multiplying the second input by 0 and adding the result to the first input, or by adding the two inputs and throwing away the result). Other relative rates are also possible by playing with Keep 1 in N and Repeat blocks, I believe. Another possible case (which would require another new block) is make signal B flow not faster than A (which apparently is the constraint that the current selector applies, with A the active input, and B an inactive input).

All this is food for thought and is probably going to take a while to mature. In more practical terms, unfortunately I now have a bunch of flowgraphs that worked in 3.10.6 (if only by sheer luck) but are broken in 3.10.7. These are like the first example I showed, but not so silly. They are digital transmitter and receiver flowgraphs where you can switch in and out at runtime different sections in order to have differential encoding or not, to have scrambling or not, to toggle between different FEC encoders/decoders, etc. This is a quite common requirement from a digital transceiver, and I don't know if there is another way in which it can be achieved, specially for the transmitter, which in my case is required to pass backpressure to its frame input. For the decoder I reckon that an alternative solution would be to have a tree of possible paths, with the signal input at the root of the tree, and 1-to-N selectors at each branches, so that only one path from the root to a leave is active at a time. The drawback is that we need to repeat blocks to form the tree, compared to the solution where we also we N-to-1 selectors.

daniestevez, 2023-08-21:
As a workaround for the flowgraphs that I have that rely on Selector consuming all the available items in inactive inputs (or the old broken behaviour), I have added to gr-satellites a modified version of the Selector block that behaves likes this. Sorry for all the spam that Github inserts because of cross-references to this issue in commit messages.