gnuradio/gnuradio #6799: Add message to var block

Bob0111, 2023-08-21:
### Feature Description

Currently, there is a block called message pair to var, but the document says input message pairs are of the form (<name> . <value>). For this block, <name> is ignored. So I think we may create a block that can just change the variable with a simple message instead of a pair. Because in this case, the <name> is ignored. So just send the pmt message to the variable will make writing the embedded python block that can change variables values much simpler.

### Feature Urgency

high (impacts broad portion of ecosystem)

Bob0111, 2023-08-21:
Input message pairs are of the form (name . value). For this block, <name> is ignored.

Animesh312, 2023-11-11:
Can I work on this issue

marcusmueller, 2023-11-12:
@Animesh312 sure! Contributions are always welcome!

So, I propose that instead of a separate "special purpose" *message to variable* block, you rather add a simple *message to dict* block, where the block has
- one parameter, which sets the first element of the message, and
- one message input, and
- one message output

The handler of the input message `msg` creates a dictionary, where the *key* is the parameter passed to the block constructor, and the value is the unmodified input `msg`. 

That way, we can use the currently existing "Message Pair to Var", and it has utility beyond just setting variable through messages.

(There is a bit of a "political" reason to why I'd rather see you spend time on a block that creates messages out of other messages than to build more "message to var" blocks: While the variable mechanism in GRC-generated python is fun for a lot of purposes, it's architecturally very questionable, and, [in some places very broken](https://github.com/gnuradio/gnuradio/issues/6766) (just one example). The general idea of messages was that it allows for much safer, and much more "obvious", modifications of parameters of blocks at runtime. However, a block like I describe above would be universally useful.)

marcusmueller, 2023-11-12:
(another benefit is that a `message_to_message_pair` block is very easy to write a unit test for!)