gnuradio/gnuradio #6800: SIN and COS Calculations are inconsistent

VladislavF, 2023-08-21:
### What happened?

The calculations for SIN and COS are not consistent depending on sample rate or even which blocks are used. See the attached flowgraph. Using a signal source set to sine or cosine and a time sink with a number of samples set to the sample rate should yield an identical result for each refresh, at least I'd imagine. This however, is only true if the sample rate is a power of two (like 32e3). 
This issue also manifests if using a complex VCO and multiply together with a frequency shift/rotator block. These are incompatible even though they should produce a mathematically identical result. using a shift->shift is fine, using a VCO+mult.->VCO+Mult. is fine, using a VCO+Mult.->shift has an inaccurate result. This is regardless of sample rate.
The more worrying thing investigating this is that the root cause likely lies with the fact that the rotator block uses VOLK for calculating sin/cos which defaults to using std::sin or std::cos whereas the analog source and VCO use this fxpt function which manually implements some sort of table lookup. These two are not consistent. Architecturally, I don't understand why there is a custom sin/cos implementation at all. Why not use VOLK/std:: everywhere?
This seems like a pretty severe issue given it's such a fundamental calculation.

As far as I can tell, this happens in any version of GR. tested with 3.10.0 and latest 3.11 main.
[sin_test.zip](https://github.com/gnuradio/gnuradio/files/12398356/sin_test.zip)

### System Information

OS: Linux/WSL
GR Installation Method: pybombs

### GNU Radio Version

3.11-git (main)

### Specific Version

3.11.0

### Steps to Reproduce the Problem

Use a frequency shift or rotator together with a complex VCO and multiply block.
Use a signal source and time sink with a sample rate set to anything except for a power of two.

willcode, 2023-08-21:
There could be cases where there is efficient SIMD method and std is slower than lookup. Getting the same value result (maybe with some noise) from the fxpt implementation is certainly a reasonable requirement.

marcusmueller, 2023-08-25:
yeah I have had a gr::math fxpt benchmark from the 3.6.5 days (someone had to shift the frequency of a target, and instead of making his Bachelor thesis actually work, he might or might not have done 1000 things around the source code), and honestly, with the compilers back then, I was not extremely impressed. This might have improved. (also, I was inexperienced in running benchmarks back then, so take with a grain of salt)

VladislavF, 2023-12-01:
Okay, so I wanted to come back to this issue because I just spent a considerable amount of time getting to the bottom of this. This has turned out to be quite the multi-faceted issue.
So I'll start by reiterating that fxpt does not in fact agree with either volk or std:: at least on ubuntu. volk and std:: do however agree exactly in output and are more consistently accurate. Though all of them can produce inconsistent results when using blocks like the signal source with a decimal frequency like 87923.234Hz or using some odd sample rate similarly.

This leads me to the next important discovery I made which is that arguments for the frequency/phase in blocks that manipulate phase or frequency MUST be doubles. Furthermore, the internal parameters that depend on this phase also must be double precision, though sin/cos can be subsequently calculated with floats.  For instance, I updated the frequency modulator, VCO, and wrote my own frequency shifter that all use doubles. I also added std:: and volk as options to the signal source. With these changes I am able to get 100% self-consistent results invariant of the frequency and sample rate. That is except for the frequency modulator block because that ends up limited by the precision of floats in the input data stream.

Now, normally you might think this is a bit pedantic and not worth fixing, but the problem is very easy to cause with irrational numbers calculated from the likes of pi. This has actually caused me many headaches in the past with phase coherence between multiple signal paths in the likes of stereo FM signal generation.

My suggestions are to firstly, replace instances of fxpt with volk or std wherever possible since these are more consistent. Next, update the relevant arguments in these blocks to be doubles. Lastly, the volk_rotate function used by the phase rotator and frequency shifters needs to be updated to do calculations with doubles internally, or I would suggest adding another phase rotator that uses doubles if this isn't possible. We did this internally, I'd be happy to share code.

Trust me when I say that people have been kicking this issue for years without even realizing it, I know I have. It's actually almost funny to see now when talking to people. Funny if it weren't infuriating to experience. 
[image]
[image]

Setup this example with the built-in rotator or mod functions and you will see resulting signal has phase drift.

willcode, 2023-12-01:
We should put some energy into fixing this up.

Do you think there would be any improvement in fxpt if `gen_sine_table.py` did proper rounding of both x and y values? And what if fxpt used a table of double instead of float? What if nbits was increased?

I'm not sure I believe the fxpt sin and sin/cos benchmark rates of 8e+13/s. I think the test is being optimized out, since the vector version is more like 6e+8/s.

Native vec is order of magnitude slower than std vec and it would be nice to avoid that hit.

VladislavF, 2023-12-01:
> Do you think there would be any improvement in fxpt if `gen_sine_table.py` did proper rounding of both x and y values? And what if fxpt used a table of double instead of float? What if nbits was increased?

That I do not know, I would have to test. Though testing should be very straightforward, it should be consistent with volk/std which both use a taylor series (on linux kernel at least). The tricky part is the mixing of python and c++ types, I've already been down this rabbit hole with #6748 with the soft decision lookup table. Python's types are not at all like c++. So mixing these is very dangerous. I ended up having to entirely rewrite that algorithm, and still the python is only consistent with cpp to within a few decimals. So this is all to say, I do not know what the correct answer is, this is a tough cookie and why I just swapped it out. I do wonder how much slower the volk implementation is than fxpt, since that should be pretty well optimized.

marcusmueller, 2023-12-01:
Re: speed differences fxpt and volk: well, the fxpt functions are inlineable, but VOLK always needs to dispatch a function call, and for single numbers, that's probably worse than the couple of multiplications and additions done inside!

willcode, 2023-12-01:
That's the speed difference between fxpt single value and fxpt vec. Thus the disbelief.
```
$ ./src/gnuradio/build/gr-blocks/tests/benchmark_nco
+——————————————————————————————————————————————————————————————————+
| fxpt sine          time: 2.7000e-07 s throughput: 7.407e+13 it/s |
| fxpt sin/cos       time: 8.2100e-07 s throughput: 2.436e+13 it/s |
| fxpt sin/cos vec   time: 5.6299e-02 s throughput: 3.552e+08 it/s |
| native sine        time: 3.6128e-01 s throughput: 5.536e+07 it/s |
| native sin/cos     time: 6.2264e-01 s throughput: 3.212e+07 it/s |
| native sin/cos vec time: 6.4556e-01 s throughput: 3.098e+07 it/s |
| basic sin/cos vec  time: 6.4627e-01 s throughput: 3.095e+07 it/s |
+——————————————————————————————————————————————————————————————————+
```
(results vary a little bit every run but same order of magnitude)

marcusmueller, 2023-12-01:
Fully get your disbelief there. Let me solidify the testing.

marcusmueller, 2023-12-01:
```
+——————————————————————————————————————————————————————————————————+
| fxpt sine          time: 8.9234e-03 s throughput: 2.241e+09 it/s |
| fxpt sin/cos       time: 2.3390e-02 s throughput: 8.551e+08 it/s |
| fxpt sin/cos vec   time: 2.8762e-02 s throughput: 6.954e+08 it/s |
| native sine        time: 1.8936e-01 s throughput: 1.056e+08 it/s |
| native sin/cos vec time: 3.9507e-01 s throughput: 5.062e+07 it/s |
| native sin/cos     time: 4.5111e-01 s throughput: 4.433e+07 it/s |
| basic sin/cos vec  time: 4.6110e-01 s throughput: 4.337e+07 it/s |
+——————————————————————————————————————————————————————————————————+
```
with proper random input and avoiding any output discard. 

(Changing to actually random input was the thing that made the difference, but a `volatile` and used sum over all calculation results should also stop smart future compilers from optimizing the loop. PR incoming.)

VladislavF, 2023-12-01:
No arguments made on performance from me, my main concern is the precision. Functions using fxpt are not self-consistent.  using a signal source with a frequency shifter does not allow returning to baseband. even using two frequency mod blocks, multiplied together does not produce a DC signal. see below another example. This should be DC/constant
[image]
[image]

VladislavF, 2023-12-01:
I only see this problem as theoretically solvable if the table can be dynamically updated or manually calculated for points that inevitably fall in between table entries, this is how most OS kernel calculations of sin/cos work afaik.

willcode, 2023-12-01:
Yah, the performance stuff was a side trip. However, 40-50 MS/s isn't going to be a great thing. Affected blocks could have a parameter for `Prefer [Precision | Performance]`.

VladislavF, 2023-12-04:
@marcusmueller Do you think it would be possible to have fxpt manually calculate values that fall in between table indices? This would be the ideal solution for performance and precision. Perhaps these could even be stored in an adjacent dict. of some sort?
@willcode you mentioned table resolution increase, I was curious what the size currently was. I haven't had a chance to really dig into that code, but what was the size? and when is it generated? I'm wondering if there are python->CPP type precision issues potentially

marcusmueller, 2023-12-04:
> @marcusmueller Do you think it would be possible to have fxpt manually calculate values that fall in between table indices? This would be the ideal solution for performance and precision. Perhaps these could even be stored in an adjacent dict. of some sort?

Yes, you'd normally use a separate look-up table lookup to get the derivative (which is easy, $\frac{\text{d}\sin(x)}{\text{d}x} = \cos(x)$, and you want both trigonometric functions anyways.), and then multiply derivative with distance to closest point in lookup table; or, you use a polynomial approximation to the derivative (so, you do the rough sin $\sin$ with a lookup table, and the derivative with a polynomial approximation), or you don't use a lookup table altogether and just go for a polynomial. 

For fun I did the latter this weekend, and was able to achieve a more precise $\sin$ on $[-\pi;\pi]$ than `numpy.sin` does, which, I might add, is different to C++'s `std::sin(float)`, and I tend to think it's wronger.

>  I was curious what the size currently was.

2048 points

> and when is it generated?

Precomputed into a header file

willcode, 2023-12-04:
I was thinking that a few more bits in the table, and either rounding or the interpolation method above, would be interesting to try out.

marcusmueller, 2023-12-05:
ah darn, I thought it was per $\pi$, so I was wrong.

willcode, 2023-12-05:
@VladislavF curious whether increasing NBITS has any effect on the error you are seeing. Try this patch if you get a chance. The API is still `float`.
```
diff --git a/gnuradio-runtime/include/gnuradio/fxpt.h b/gnuradio-runtime/include/gnuradio/fxpt.h
index d8eaca28c..ddb15ff97 100644
--- a/gnuradio-runtime/include/gnuradio/fxpt.h
+++ b/gnuradio-runtime/include/gnuradio/fxpt.h
@@ -30,7 +30,7 @@ namespace gr {
 class GR_RUNTIME_API fxpt
 {
     static constexpr int WORDBITS = 32;
-    static constexpr int NBITS = 10;
+    static constexpr int NBITS = 16;
     static constexpr uint32_t ACCUM_MASK = ((1 << (WORDBITS - NBITS)) - 1);
     static const float s_sine_table[1 << NBITS][2];
     static const float PI;
diff --git a/gnuradio-runtime/lib/math/gen_sine_table.py b/gnuradio-runtime/lib/math/gen_sine_table.py
index b45cb52e7..c9b6a2453 100644
--- a/gnuradio-runtime/lib/math/gen_sine_table.py
+++ b/gnuradio-runtime/lib/math/gen_sine_table.py
@@ -33,7 +33,7 @@ def scaled_sine(x):
 
 
 def gen_sine_table():
-    nbits = 10
+    nbits = 16
     nentries = 2**nbits
 
     # min_x = -2**31
```

VladislavF, 2023-12-05:
Sure @willcode I'll give it a shot

willcode, 2023-12-05:
The table stores y=f(int(x)) and m, then the lookup does y + dx * m or something similar, for interpolation. Agree that the "whole" solution is not just to increase NBITS. Wondering if it's part of the solution.

ValeZAA, 2024-01-22:
Tbis looks very relevant, I mean to the issue linked 

https://stackoverflow.com/questions/4633177/c-how-to-wrap-a-float-to-the-interval-pi-pi/29871193

marcusmueller, 2024-01-22:
@ValeZAA we're already doing something similar (smarter, actually)

argilo, 2024-10-01:
I had a look into this, and it seems like the problems reported here and in #7208 are due to phase accumulator error. The Signal Source and VCO blocks use a 32-bit fixed-point phase accumulator, while other blocks (Rotator, Frequency Shift, Frequency Xlating FIR Filter, etc.) use a single-precision complex phase accumulator (which has lower precision).

Since there are two types of phase accumulators, errors are readily visible when their outputs are compared. Switching to the same type of accumulator everywhere would make it harder to observe the error, but it would still be there nonetheless.

Errors due to arithmetic precision are unavoidable, so I'm not sure there's anything GNU Radio can do about this. A flow graph should not depend on phase accumulators being free of error.

VladislavF, 2024-10-01:
Sure, I would argue it should be consistent. Obviously no digital value will ever be perfectly accurate and perfectly precise. This issue is most obvious when the implementations are inconsistent and/or when they lack reasonable precision in the first place. When I fixed this internally, I remember simply updating the functions in the VCO and signal sources to use std::cos/sin and changed the phase accumulation to a double. Granted we then also wrote our own frequency shifter that used doubles as well. In my mind, we should just try to make things consistent if we can. That way this doesn't continue to cause unexpected issues like FM stereo not working correctly due to frequency translator blocks and signal sources not being consistent.

VladislavF, 2024-10-01:
When I get a chance I will see if I can find any of the code where I had this mostly resolved

argilo, 2024-10-01:
I agree that switching to double would gain something, since the higher precision would reduce the error magnitude. But error would of course still be present, and the switch would come at the cost of performance.

Consistency across blocks is debatable. I would argue that if a flow graph depends on arithmetic error being "just so", then the fault lies with the flow graph. A consistent phase accumulator implementation might let you get away with more bugs in your flow graph, but that's not necessarily a good thing.

VladislavF, 2024-10-01:
I think on many modern machines float and double have about the same performance, but I could be wrong.

What is a proper way to implement the FM stereo signal generation that doesn't need consistency between the pilot tone and the frequency translator? Currently, I use the digital equivalent of an LO doubler which is quite inefficient computationally. I'm sure there are other applications, but let's stick with this example for now.

argilo, 2024-10-01:
> I think on many modern machines float and double have about the same performance, but I could be wrong.

I think that's kinda true on x86 (although I'm not sure it carries over to SIMD) but not true on ARM.

> Currently, I use the digital equivalent of an LO doubler which is quite inefficient computationally.

Yes, I'd say generating a 19 kHz tone and squaring it to get 38 kHz is the right approach. Phase accuracy is critical in FM stereo, so generating the tones independently seems risky.

argilo, 2024-10-01:
> generating the tones independently seems risky

Out of curiosity, I had a peek at gr-rds and that's precisely what it does (with a pair of Signal Source blocks). It's only by chance that it works, because the sample rate (380 kHz) happens to be an exact multiple of 19 kHz. That should probably be fixed.

argilo, 2024-10-01:
Switching to a double-precision phase accumulator wouldn't solve the problem for FM stereo, since two double-precision phase accumulators would still drift apart if not for a fortunate choice of sample rate.

argilo, 2024-10-02:
I proposed a fix for gr-rds: https://github.com/bastibl/gr-rds/pull/79

Its RDS subcarrier was in fact drifting relative to the pilot signal (contrary to the RDS standards), because its frequency does not evenly divide into the sample rate.