gnuradio/gnuradio #6809: Repeat block: buffer overflow for large interpolations

eltos, 2023-08-28:
### What happened?

Consider this simple flow graph:
[image]

When changing the value of repeat from 1k to 10k at runtime the data flow stops with the following error:

    thread_body_wrapper :error: ERROR thread[thread-per-block[1]: <block repeat(2)>]: Buffer too small for min_noutput_items

However, if the repeat is already set to 10k by default when the flow graph starts, sufficient buffer is allocated and no error occurs.

I consider this a bug because there is no fundamental reason for the repeat block to allocate such a large buffer in the first place. **Instead of exceeding the allocated buffer size, the block should put back-pressure on the upstream block.** Also, the same functionality can be realized with two subsequent repeat blocks without such issues (limiting the granularity of possible interpolation of course):
[image]

### System Information

OS: Linux Ubuntu
GR Installation Method: Build from source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

See above

willcode, 2023-08-28:
Output buffers are allocated for each block when the flowgraph starts, and are not reallocated during a run. An unfortunate effect of this is that when a block decides it needs a larger output buffer due to a change in state (interpolation changes in this case) during a run, the output buffer size could be too small.

The min buffer size can be set for each block (see the advanced tab) if the output buffer size requirements are known.

Backpressure is being used here, it's just that the Repeat block tries to output "Interpolation" samples at a time. One possible change would be for the block to remember the last item it was replicating and to continue repeating that item across work calls if the output buffer is full. That would be a fairly simple change.

With respect to "allocate such a large buffer", I don't see anything wrong with how the block is implemented. The problem happens when the buffer is too small.

willcode, 2023-08-28:
Feature request: modify Repeat block so interpolation can continue across work calls when output buffer is to small, due to a change in Interpolation after a run starts.

closed 2023-08-29 by willcode