gnuradio/gnuradio #6816: newmod still depends on cppunit, though GNU Radio moved on to Boost::test

marcusmueller, 2023-08-30:
### What happened?

While checking something, I realized CI still needs cppunit, because our gr_modtool template uses it. But GNU Radio has long moved on.

### System Information

--

### GNU Radio Version

3.11-git (main)

### Specific Version

main

### Steps to Reproduce the Problem

git grep cppunit

dl1ksv, 2023-08-31:
At the moment cppunit may be required as 'gr_modtool add' still claims to support 3.7

https://github.com/gnuradio/gnuradio/blob/a05e13ac9b5aefaee42e585b246bffab05b64bc8/gr-utils/modtool/core/add.py#L230-L239

I wonder if it makes sense to support oot modules with version < 3.8 in gnuradio 3.11

marcusmueller, 2023-08-31:
You're right. This should be a different ticket!