gnuradio/gnuradio #6820: gr-fec's RS does access to out-of-bounds offset

marcusmueller, 2023-09-03:
### What happened?

Happily read the debian 12 CI output, and spotted the below warning about a memcpy

### System Information

OS: debian 12 x86_64 CI container

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

build on debian 12 CI

### Relevant log output

```shell
2023-09-02T13:33:38.1123415Z In file included from /__w/gnuradio/gnuradio/gr-fec/lib/reed-solomon/char.c:8:
2023-09-02T13:33:38.1124514Z /__w/gnuradio/gnuradio/gr-fec/lib/reed-solomon/decode_rs.h: In function 'decode_rs_char':
2023-09-02T13:33:38.1150883Z /__w/gnuradio/gnuradio/gr-fec/lib/reed-solomon/decode_rs.h:183:5: warning: 'memcpy' forming offset 4294967295 is out of the bounds [0, 4294967295] [-Warray-bounds]
2023-09-02T13:33:38.1151550Z   183 |     memcpy(&reg[1], &lambda[1], NROOTS * sizeof(reg[0]));
2023-09-02T13:33:38.1152260Z       |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

marcusmueller, 2023-09-03:
hm, gen_ccsds.c correctly uses `unsigned char` for the index arrays, but the generated ccsds_tab file uses the implementation-signedness-defined `char`; wonder what the reason was!

daniestevez, 2023-09-03:
I think the arrays in `ccsds_tab.c` should be `unsigned char`. No idea why the are `signed char` at the moment (I can't remember if this is something I touch or copied from Phil's code). However, I don't think this matters, though. They are declared as `extern unsigned char` in `fixed.h`, which I think is the only place where they're used. The signedness of `CCSDS_alpha_to`, which is used via `ALPHATO` does not seem to matter, but the signedness of `CCSDS_index_of`, which is used via `INDEX_OF` does seem to matter. So it's probably better to change them to `unsigned char`.

Nevertheless, I don't understand how this is related to the `memcpy()` that produces the warning, since none of this seems to be used there (`decode_rs_char` is not even supposed to use these CCSDS arrays). I'm not understanding the memcpy warning. The only slightly unusual thing is that `NROOTS` isn't actually a constant, but rather `rs->nroots`, which is a `unsigned int`. `reg` and `lambda` are `unsigned char` arrays of size `rs->nroots` (yes, dynamically sized arrays on the stack). I guess technically speaking `rs->nroots` could be `0` or `1` and then `&reg[1]` and `&lambda[1]` would be UB, but in practice I hope (and assume) that `rs->nroots` is set to a sensible value when this runs. I'm also assuming that `sizeof(reg[0]) == sizeof(unsigned char)` even though `reg[0]` might not exist.

marcusmueller, 2023-09-04:
hmmm, yeah seems I really need to debug this, might be a false compiler alarm

marcusmueller, 2024-02-14:
really seems to be a false compiler diagnostic. Can't really preclude that there's still something wrong, but I don't think this is a symptom.

closed 2024-02-14 by marcusmueller