gnuradio/gnuradio #6826: UDP source block (since 3.9) works weird with "Vector Length" parameter

shinchik17, 2023-09-11:
### What happened?

New UDP Source block (which were added into GR since 3.9) doesn't work with vector length set more than 5. It just doesn't produce any samples at all. In comparison, old UDP Source block (I checked it in GR 3.8.2) worked fine with both vector length set to 1 (stream mode) and with arbitrary vector length.

### System Information

OS: Windows 10
GR Installation Method: radioconda installer

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

To reproduce the problem you need to check attached flowgraphs. They art:
#1 udp_transmit310.grc - flowgraph for sending data to udp 
#2 udp_receive310.grc - flowgraph for receiving data from udp
I tried both transmitting between 2 computers and sending data to localhost, there is no difference, that's on your choice.
#3 udp_receive38.grc - GR 3.8.2 flowgraph for receiving data from udp. You may use it for comparison. There are 2 UDP source blocks there with stream mode and with vector length > 0 and both work fine.

So. You need to set suitable IP's and ports into flowgraphs (at least #1 и #2). Then run #1 and  #2. Try to see data in #2 flowgraph using differen vector length value in UDP Source block. For me, the workable value is 1-4, then block just stops producing anything to its output. (Of course values 2-4 damages the data but at least data is receiving). Why can't I receive entire packet (1472 length in example) into 1 vector with the same length? Idk, but it doesn't work :) 

[udp_troubleshooting.zip](https://github.com/gnuradio/gnuradio/files/12575800/udp_troubleshooting.zip)

willcode, 2023-09-11:
For debugging, could you set Minoutbuf under the Advanced tab to something larger, like 256*1024 and see if that changes the behavior at all?

willcode, 2023-09-11:
I tried out your examples, and the results are definitely unreliable with larger vector sizes. Using a larger min output buffer helps somewhat, but there are still drops. Something is wrong in the block logic here.

Of course, you can just move the vectorization down a block, e.g.,
[image]
so do you really need to get vectors out of the UDP source?

shinchik17, 2023-09-12:
Of course, I need it. If I don't, I wouldn't write this :) The problem is that I have custom header inside my packets and data bitrate is over 3 Gbit/s so data dropping is inevitable. That's why I need to be sure that header is really located at the beginning (allocating 1 packet into 1 vector garantees that). When I set vector length to 1 (stream mode), strange things happen sometimes: like header moves from beginning to another place which causes incorrect header exclusion by block "Keep M in N"  which causes incorrect data interpretation.

To be honest, I absolutely don't understand what was the reason and what are the advantages of replacing old udp blocks to these ones. Old one worked absolutely fine and logical. New block: 
- absence of IP adress field filter, which adds some other trash packets to your socket if you work in some network;
- doesn't support vector length field.
Advantages? I didn't noticed any. I even compared perfomance with 3.8.2 block - there is no difference (at least on my tasks).

So I'm really looking forward to solution of this problem :)

willcode, 2023-09-12:
In your example, item size is 8, vector length is 1472 // 8, so the output multiple comes out as 1 and the code below tries to make it 2. This may be what's happening. Will look more later.
```
    d_block_size = d_itemsize * d_veclen;

...

    int out_multiple = (d_payloadsize - d_header_size) / d_block_size;

    if (out_multiple == 1)
        out_multiple = 2; // Ensure we get pairs, for instance complex -> ichar pairs

    gr::block::set_output_multiple(out_multiple);
```

willcode, 2023-09-12:
@shinchik17 give this patch a try. Basically, the veclen was not being included correctly in calculations.
The vector size needs to divide evenly into the payload size. I can't think of a use for thinking in terms of vectors, but allowing them to span packets. Probably something we should handle gracefully.
```
$ factor 1472
1472: 2 2 2 2 2 2 23
```
using header=None. So veclen can be 1, 2, 4, 8, 16, 23, 46, 64, 92, 184.
```
diff --git a/gr-network/lib/udp_source_impl.cc b/gr-network/lib/udp_source_impl.cc
index 2fddee101..4b259a700 100644
--- a/gr-network/lib/udp_source_impl.cc
+++ b/gr-network/lib/udp_source_impl.cc
@@ -100,9 +100,9 @@ udp_source_impl::udp_source_impl(size_t itemsize,
     }
 
     d_precomp_data_size = d_payloadsize - d_header_size;
-    d_precomp_data_over_item_size = d_precomp_data_size / d_itemsize;
+    d_precomp_data_over_item_size = d_precomp_data_size / d_block_size;
 
-    int out_multiple = (d_payloadsize - d_header_size) / d_block_size;
+    int out_multiple = d_precomp_data_over_item_size;
 
     if (out_multiple == 1)
         out_multiple = 2; // Ensure we get pairs, for instance complex -> ichar pairs
```

willcode, 2023-09-12:
@marcusmueller @argilo  could you take a look at this logic in general? My eyes are getting blurry looking at it. It seems that "casting" the received data as vectors hasn't been used much.

shinchik17, 2023-09-13:
> @shinchik17 give this patch a try. Basically, the veclen was not being included correctly in calculations. The vector size needs to divide evenly into the payload size. I can't think of a use for [...]

I understand that "The **vector size** needs to divide evenly into the payload size", that's why I set **it** to184 in my examples. I don't understand how I can try your path. I use radioconda version of GR on Windows as I noticed above. Are there any ways to change the source code of block? What should my steps be?

willcode, 2023-09-13:
Using radioconda, you'd have to wait for the next release. This can probably make the Oct release.

PR #6835 (2023-09-18): network: improved UDP Source vectorization
## Description
The calculation for set_output_multiple() in did not take vectors into account correctly. Use of vectors is still tricky, as the vector size needs to divide into the payload size evenly. Incorrect values result in corrupted data.

## Related Issue
Fixes #6826

## Which blocks/areas does this affect?
UDP Source

## Testing Done
Sample GRCs provided with issue.

## Checklist
- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significan [... cut]
closed 2023-09-18 by willcode