gnuradio/gnuradio #6828: Make Test failure

kb6ibb, 2023-09-11:
### What happened?

I understand from the documentation that not all of the "make test", tests will pass.   However, there is a failure with what appears to be a memory allocation in the qt opengl that may need attention.     

### System Information

Linux:  Rocky 9.2
GR Installation Method:  Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-520-ga05e13ac (Python 3.9.16)

### Steps to Reproduce the Problem

make test

### Relevant log output

```shell
======================================================================
ERROR: test02 (__main__.test_qtgui_opengl)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/kb6ibb/src/Gnuradio-3.11/gnuradio/gr-qtgui/python/qtgui/qa_qtgui_opengl.py", line 27, in test02
    self.qtsnk = qtgui.FosphorGlSink(1, 16, 512, 64, 1.0, 0.1, 0.99, 16.0, 1024.0)
  File "/usr/local/lib64/python3.9/site-packages/gnuradio/qtgui/fosphorgl_sink.py", line 47, in __init__
    self.fosphor_sink = qtgui.fosphor_display(
MemoryError: std::bad_alloc

----------------------------------------------------------------------
Ran 2 tests in 0.079s

FAILED (errors=1)
<end of output>
Test time =   0.27 sec
----------------------------------------------------------
Test Failed.
"qa_qtgui_opengl" end time: Sep 09 22:51 CDT
"qa_qtgui_opengl" time elapsed: 00:00:00
----------------------------------------------------------
```

willcode, 2023-09-11:
Thanks, I linked that to the original PR.

argilo, 2023-12-10:
@dl1ksv Is this now fixed?

dl1ksv, 2023-12-10:
@argilo Yes the std::bad_alloc is fixed.

argilo, 2023-12-10:
Thanks for confirming. I'll close this off.

closed 2023-12-10 by argilo