gnuradio/gnuradio #6830: Error while building gnuradio source code --> error: ‘RECV_MODE_ONE_PACKET’ is not a member of ‘uhd::device

markibonge, 2023-09-11:
### What happened?

A bug happened! While I was trying to build the gnu radio source code on my ubuntu 18.04.06

Below is the error message I kept getting. I looked up and it seems like there were similar issues in the past. But I couldn't find a solution for the specific error message below. Please let me know what I can do to solve this issue.

/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
template class auto_ptr;
^~~~~~~~
In file included from /home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.h:24:0,
from /home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:24:
/home/opnear/workarea/gnuradio/gr-uhd/include/gnuradio/uhd/usrp_source.h:31:19: error: redefinition of ‘struct uhd::stream_args_t’
struct GR_UHD_API stream_args_t {
^~~~~~~~~~~~~
In file included from /home/opnear/workarea/gnuradio/gr-uhd/lib/gr_uhd_common.h:26:0,
from /home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:23:
/usr/local/include/uhd/stream.hpp:47:16: note: previous definition of ‘struct uhd::stream_args_t’
struct UHD_API stream_args_t
^~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In constructor ‘gr::uhd::usrp_source_impl::usrp_source_impl(const uhd::device_addr_t&, const uhd::stream_args_t&, bool)’:
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:74:7: error: class ‘gr::uhd::usrp_source_impl’ does not have any field named ‘_recv_timeout’
_recv_timeout(0.1), // seconds
^~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:75:7: error: class ‘gr::uhd::usrp_source_impl’ does not have any field named ‘_recv_one_packet’
_recv_one_packet(true),
^~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In member function ‘virtual void gr::uhd::usrp_source_impl::set_recv_timeout(double, bool)’:
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:464:5: error: ‘_recv_timeout’ was not declared in this scope
_recv_timeout = timeout;
^~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:464:5: note: suggested alternative: ‘set_recv_timeout’
_recv_timeout = timeout;
^~~~~~~~~~~~~
set_recv_timeout
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:465:5: error: ‘_recv_one_packet’ was not declared in this scope
_recv_one_packet = one_packet;
^~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:465:5: note: suggested alternative: ‘one_packet’
_recv_one_packet = one_packet;
^~~~~~~~~~~~~~~~
one_packet
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In member function ‘void gr::uhd::usrp_source_impl::flush()’:
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:512:29: error: ‘using element_type = class uhd::device {aka class uhd::device}’ has no member named ‘recv’
_dev->get_device()->recv(outputs,
^~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:516:49: error: ‘RECV_MODE_FULL_BUFF’ is not a member of ‘uhd::device’
::uhd::device::RECV_MODE_FULL_BUFF,
^~~~~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In member function ‘virtual int gr::uhd::usrp_source_impl::work(int, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:600:44: error: ‘using element_type = class uhd::device {aka class uhd::device}’ has no member named ‘recv’
size_t num_samps = _dev->get_device()->recv(output_items,
^~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:604:64: error: ‘RECV_MODE_FULL_BUFF’ is not a member of ‘uhd::device’
::uhd::device::RECV_MODE_FULL_BUFF,
^~~~~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:608:41: error: ‘using element_type = class uhd::device {aka class uhd::device}’ has no member named ‘recv’
num_samps = _dev->get_device()->recv(output_items,
^~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:612:61: error: ‘RECV_MODE_ONE_PACKET’ is not a member of ‘uhd::device’
::uhd::device::RECV_MODE_ONE_PACKET,
^~~~~~~~~~~~~~~~~~~~
gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/build.make:86: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o' failed
make[2]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o] Error 1
CMakeFiles/Makefile2:14760: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all' failed
make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

### System Information

[... 91 lines cut ...]
         _dev->get_device()->recv(outputs,
                             ^~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:516:49: error: ‘RECV_MODE_FULL_BUFF’ is not a member of ‘uhd::device’
                                  ::uhd::device::RECV_MODE_FULL_BUFF,
                                                 ^~~~~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc: In member function ‘virtual int gr::uhd::usrp_source_impl::work(int, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:600:44: error: ‘using element_type = class uhd::device {aka class uhd::device}’ has no member named ‘recv’
     size_t num_samps = _dev->get_device()->recv(output_items,
                                            ^~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:604:64: error: ‘RECV_MODE_FULL_BUFF’ is not a member of ‘uhd::device’
                                                 ::uhd::device::RECV_MODE_FULL_BUFF,
                                                                ^~~~~~~~~~~~~~~~~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:608:41: error: ‘using element_type = class uhd::device {aka class uhd::device}’ has no member named ‘recv’
         num_samps = _dev->get_device()->recv(output_items,
                                         ^~~~
/home/opnear/workarea/gnuradio/gr-uhd/lib/usrp_source_impl.cc:612:61: error: ‘RECV_MODE_ONE_PACKET’ is not a member of ‘uhd::device’
                                              ::uhd::device::RECV_MODE_ONE_PACKET,
                                                             ^~~~~~~~~~~~~~~~~~~~
gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/build.make:86: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o' failed
make[2]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/usrp_source_impl.cc.o] Error 1
CMakeFiles/Makefile2:14760: recipe for target 'gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all' failed
make[1]: *** [gr-uhd/lib/CMakeFiles/gnuradio-uhd.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```

willcode, 2023-09-11:
Why are you including the instruction on building v3.7?

markibonge, 2023-09-11:
perhaps I'm looking at the wrong instruction. could you please the link on
the instructions of building the newer version:?

willcode, 2023-09-11:
https://wiki.gnuradio.org/index.php?title=InstallingGR

closed 2023-09-11 by willcode