gnuradio/gnuradio #6863: Improved ZMQ Teardown

SaikWolf, 2023-09-23:
### Feature Description

For more complex usage, the ZMQ sources/sinks can hang when trying to shutdown/stop the flowgraphs they're in.

Making sure the destructor properly terminates resources should help fix that.

(Most notable within Python usage)

### Feature Urgency

medium (would be nice to have in the near future)

### More Information

A quick patch for my own use is here: https://github.com/SaikWolf/gnuradio/pull/2

I'm sure I'm missing things needed for full integration, but if anyone finds it useful as a starting point.

willcode, 2023-09-25:
That looks good. Do you want to submit that against `main`?

SaikWolf, 2023-09-25:
Anywhere that makes it simple to integrate would be fine.

willcode, 2023-09-25:
You can either submit a PR against our `main`, or I can pull this from your repo.

vmw, 2023-10-10:
This commit results in a compilation error when running under Oracle 8. This compilation failure is reproducible using the following Dockerfile:

https://github.com/pervices/pvpkg/blob/f4c6410/Oracle/Oracle8/GnuRadio/Dockerfile

We specifically get the following compilation error:

---

[ 31%] Building CXX object gr-zeromq/lib/CMakeFiles/gnuradio-zeromq.dir/base_impl.cc.o
cd /root/rpmbuild/BUILD/gnuradio/build/gr-zeromq/lib && /opt/rh/gcc-toolset-11/root/usr/bin/c++ -DFMT_LOCALE -DFMT_SHARED -DGR_MPLIB_GMP -DGR_PERFORMANCE_COUNTERS -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -Dgnuradio_zeromq_EXPORTS -I/root/rpmbuild/BUILD/gnuradio/gr-zeromq/lib/../include -I/root/rpmbuild/BUILD/gnuradio/gnuradio-runtime/lib/../include -I/root/rpmbuild/BUILD/gnuradio/build/gnuradio-runtime/lib/../include -I/root/rpmbuild/BUILD/gnuradio/gnuradio-runtime/lib/pmt/../../include -I/root/rpmbuild/BUILD/gnuradio/gnuradio-runtime/include -I/root/rpmbuild/BUILD/gnuradio/build/gnuradio-runtime/include -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches   -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fvisibility=hidden -Wsign-compare -Wno-uninitialized -fcx-limited-range -DNDEBUG -fPIC -std=c++17 -Winvalid-pch -include /root/rpmbuild/BUILD/gnuradio/build/gr-zeromq/lib/CMakeFiles/gnuradio-zeromq.dir/cmake_pch.hxx -MD -MT gr-zeromq/lib/CMakeFiles/gnuradio-zeromq.dir/base_impl.cc.o -MF CMakeFiles/gnuradio-zeromq.dir/base_impl.cc.o.d -o CMakeFiles/gnuradio-zeromq.dir/base_impl.cc.o -c /root/rpmbuild/BUILD/gnuradio/gr-zeromq/lib/base_impl.cc
/root/rpmbuild/BUILD/gnuradio/gr-zeromq/lib/base_impl.cc: In destructor 'virtual gr::zeromq::base_impl::~base_impl()':
/root/rpmbuild/BUILD/gnuradio/gr-zeromq/lib/base_impl.cc:90:15: error: 'class zmq::context_t' has no member named 'shutdown'
   90 |     d_context.shutdown();
      |               ^~~~~~~~
make[2]: *** [gr-zeromq/lib/CMakeFiles/gnuradio-zeromq.dir/build.make:96: gr-zeromq/lib/CMakeFiles/gnuradio-zeromq.dir/base_impl.cc.o] Error 1
make[2]: Leaving directory '/root/rpmbuild/BUILD/gnuradio/build'
make[1]: *** [CMakeFiles/Makefile2:8274: gr-zeromq/lib/CMakeFiles/gnuradio-zeromq.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Entering directory '/root/rpmbuild/BUILD/gnuradio/build'

---

This error is observed with gcc-11

---

[root@cross build]# /opt/rh/gcc-toolset-11/root/usr/bin/c++ -v
Using built-in specs.
COLLECT_GCC=/opt/rh/gcc-toolset-11/root/usr/bin/c++
COLLECT_LTO_WRAPPER=/opt/rh/gcc-toolset-11/root/usr/libexec/gcc/x86_64-redhat-linux/11/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,lto --prefix=/opt/rh/gcc-toolset-11/root/usr --mandir=/opt/rh/gcc-toolset-11/root/usr/share/man --infodir=/opt/rh/gcc-toolset-11/root/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl=/builddir/build/BUILD/gcc-11.2.1-20220127/obj-x86_64-redhat-linux/isl-install --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=x86-64 --build=x86_64-redhat-linux
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.1 20220127 (Red Hat 11.2.1-9.1.0.6) (GCC) 

---

We are presently working around this issue by disabling zeromq when compiling for Oracle8.6, using the  -DENABLE_GR_ZEROMQ=OFF cmake flag.

willcode, 2023-10-10:
cppzmq fixed the missing function in 4.7.0. Are you able to upgrade cppzmq?

vmw, 2023-10-10:
Unfortunately, using the default redhat/oracle ol8_developer_EPEL, the only supported version of cppzmq is 4.4.1.

---
[]# yum info cppzmq
Last metadata expiration check: 1:40:51 ago on Tue Oct 10 12:26:57 2023.
Available Packages
Name         : cppzmq
Version      : 4.4.1
Release      : 1.el8
Architecture : src
Size         : 43 k
Source       : None
Repository   : ol8_developer_EPEL
Summary      : Header-only C++ binding for libzmq
URL          : https://zeromq.org
License      : MIT

---

It doesn't look like there's an easy way to install this on RH/Oracle - I think you either need to introduce a hard version requirement, which doesn't seem to be desirable for a maint- branch, or else additional code to check which version of cppzmq is installed, and then use the ifdefs as appropriate.

marcusmueller, 2023-10-10:
cppzmq is really just a single header file, so luckily it's not hard to manually install. However, that's not what I'd consider a *good* option, just a very viable workaround

willcode, 2023-10-10:
We might need to ifdef the shutdown calls. @SaikWolf can you take a look at this?

vmw, 2023-10-10:
> cppzmq is really just a single header file, so luckily it's not hard to manually install. However, that's not what I'd consider a _good_ option, just a very viable workaround

Unfortunately, the primary users of our Oracle/RH versions include customer that have high assurance requirements. These installations typically have STIGS (Security Technical Implementation Guides (STIGs) and extensive software that ensures the integrity of system libraries. Unless we're prepared to maintain and support the updated cppzmq package (which we are not), we are unable to just manually overwrite the system library without actually going through the proper installation process.

willcode, 2023-10-10:
@vmw Since this was a bug (omission of a function) in the older version of the bindings, and RHEL/derivatives have chosen not to update for v8, I think we'll leave this as is on our end. For your build, you can place the `zmq.hpp` file in the GNU Radio source tree and or modify a makefile to use that version. Hop on the General Chat or Development channels on matrix if you have any trouble doing that.

willcode, 2023-10-10:
Original issue was addressed by #6865 but didn't auto close. Closing now. Further discussion can be in a new issue or PR.

SaikWolf, 2023-10-10:
Hmm, there's probably an ifdef that would work to call the czmq shutdown directly for older cppzmq versions.

willcode, 2023-10-10:
> Hmm, there's probably an ifdef that would work to call the czmq shutdown directly for older cppzmq versions.

I think we'll skip that at this point. Thanks for responding.

vmw, 2023-10-10:
@willcode 

> chosen not to update for v8, I think we'll leave this as is on our end. For your build, you can place the `zmq.hpp` file in the GNU Radio source tree and or modify a makefile to use that version.

I understand the rationale, and what follows is a strong opinion weakly held: I don't think introducing a commit that breaks compilation for a distribution that includes long term support through 2029 without a better form of warning is particularly friendly. At minimum, I think the CMakeLists should be updated to specify the minimum required version to provide an informative error, if only to make it easier to identify this as the problem when users try and compile it.

Again, given the constraints of the STIGs and the nature of our own support, we've just entirely disabled the zeromq functionality for now, but a warning/error message is a much nicer way forward if the issue isn't going to be fixed.

willcode, 2023-10-10:
Completely understand your situation and perspective!

Adding a version check to our `FindZeroMQ.cmake` would be a good idea, so people don't have to wait until a bad compile to find the mismatch. I'll look into it. Our current check is just for the existance of `zmq.hpp` and nothing more.

As far as LTS OSs, we won't guarantee compatibility with any distro just because they "support" it until the end of time. I'd actually say that good "support" includes updating packages periodically, at least where backward compatibility can be maintained. Please let RHEL/Oracle/Copycats know that cppzmq needs an update!

marcusmueller, 2023-10-11:
I'll point out that you're a commercial consumer of that long-time stability. It would be very welcome if you contributed the CMake code to support that!