gnuradio/gnuradio #6864: Improved Error Messages for PMT Deserialization in Python

SaikWolf, 2023-09-23:
### Feature Description

The current exposed function for PMT deserialization on the Python side is `deserialize_str()` which expects `bytes` rather than a `str`. Just adding a check for it on the C++ side that instead of saying `tag = 194` the error message says something like "use `bytes` not `str`"

### Feature Urgency

low (just an idea)

### More Information

A quick (untested) patch in this direction is here: https://github.com/SaikWolf/gnuradio/pull/1

If it helps anyone.

willcode, 2023-09-25:
Is the `0xc2` a Python constant? I'm just not sure that we should, in the general C++ code, we should be making assumptions about the source of the unknown tag based on a single byte.

SaikWolf, 2023-09-25:
Any time I sent in a `str` that was the constant `invalid tag` that kept being reported. I haven't dug far enough to confirm that without doubt.

All I'm certain of is that this caused too much time in debug before it was clear that was the problem, so having some kind of catch  with useful warning would be extremely useful, but `invalid tag: 194` leaves too much confusion.

willcode, 2023-09-25:
The first thing I tried gave an different error. I don't think we can try to guess why the user data is bad.
```
>>> pmt.deserialize_str(str(pmt.serialize_str(pmt.to_pmt('hello'))))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: pmt::deserialize: malformed input stream, tag value = : 98
```