gnuradio/gnuradio #6869: gr-qtgui: QT GUI Msg Push Button add callback for Message Value

duggabe, 2023-09-26:
### Feature Description

Currently the QT GUI Msg Push Button sends a fixed 'Message Value' when the button is clicked. It would be very useful to have a Variable name in that  'Message Value' field so that some other block variable or a calculation of values could be sent it the message.

### Feature Urgency

medium (would be nice to have in the near future)

### More Information

The fact that the PMT value does not change has been tested with several different flowgraphs on GNU Radio versions 3.10.4.0, 3.10.6.0, and 3.11.0.0git

This was initially observed by @Akira25

dl1ksv, 2023-10-13:
@duggabe Maybe I misunderstand something, but you can already use a variable in the 'Message Value' field.

I used a simple flowgraph with a Msg Push Button connected to a Message Debug block and a variable block named msg.

I tested different values for msg
3*4
3.1*4.2
'3*4'

If you adjust the type  accordingly, the button sends the expected result

duggabe, 2023-10-13:
@dl1ksv The problem which triggered this issue was using a QT GUI Chooser and a Msg Push Button. When the user selected a different option, the Push Button did not change value from the one at the time the flowgraph was created.

dl1ksv, 2023-10-15:
@duggabe Thanks for clarification. I provided a possible solution.

PR #6916 (2023-10-28): gr-qtgui: QT GUI Msg Push Button add callback for Message Value

## Description

Add a callback for Message callback

Fixes #6869

## Which blocks/areas does this affect?

qtgui_msgpushbutton
## Testing Done

Use a flowgraph containing a QT GUI Chooser to change the message value of a QT GUI Msg Push Button and display the results .

## Checklist
commit d362942a (2023-10-28): gr-qtgui: QT GUI Msg Push Button add callback for Message Value

Add a callback for Message callback

Fixes #6869
commit 9288d06f (2023-12-10): gr-qtgui: QT GUI Msg Push Button add callback for Message Value

Add a callback for Message callback

Fixes #6869

(cherry picked from commit d362942a2eb025c0fbcee506bfcd782be87fc779)
closed 2023-10-28 by willcode