gnuradio/gnuradio #6891: OFDM Loopback Example drops packets even with 0 noise in simulation

bradleeharr, 2023-10-07:
### What happened?

A bug happened!
Running the [OFDM Loopback example](https://github.com/gnuradio/gnuradio/blob/master/gr-digital/examples/ofdm/ofdm_loopback.grc), when changing the vector source to anything other than range(0, packet_len), and adding a time sink to compare the input and output streams, I can see that packets are dropped even when there is no noise in the simulation. 
The modified flowgraph: [ofdm_loopback (1).grc.txt](https://github.com/gnuradio/gnuradio/files/12836791/ofdm_loopback.1.grc.txt)

In this image below, you can see one example. The input is a ramp from 0 to 255. The received signal gets shifted up by 50 (the packet length), indicating that 50 samples were not received and instead it jumped to the next packet. 

[image]

I understand that packets may be dropped when there is noise and channel effects, but this behavior occurs even in the absence of any noise or channel model in the simulation, which leads me to suspect there might be an issue either with the OFDM receiver or transmitter, or possibly with the documentation or the example flowgraph itself. 

### System Information

OS: Windows
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.6.0

### Steps to Reproduce the Problem

Run my flowgraph: 
[ofdm_loopback (1).grc.txt](https://github.com/gnuradio/gnuradio/files/12836791/ofdm_loopback.1.grc.txt)

marcusmueller, 2024-02-14:
I replaced your two-input Time sink with a subtraction to just see for how long the difference stays constant. I also completely removed the channel model, instead of just setting it to zero-effect.

We're losing a packet every couple of packets. This is a bug, and it's a rather serious one.

Observations:

1. hypothesis: depends on rate, as OFDM RX needs time to asynchronously from the header decoding branch back into the header/paylod demux. **nope**, happens at very low sampling rates just as often
2. hypothesis: multithreading woes due to the shared equalizer object, but that doesn't seem to be the case anymore

blancos-dev, 2024-11-18:
Hi, I would like to know if this issue still persists in the latest GNU Radio versions. Has anyone found a workaround or solution for this packet dropping problem in the OFDM Loopback example?

I'm interested in knowing:
1. If this has been fixed in newer release 3.10.10
2. If there are any known workarounds for those who need reliable OFDM packet transmission
3. If additional debugging or testing would be helpful

Thanks in advance!

marcusmueller, 2024-11-19:
Please use the file attached to the original bug report to check, @blancos-dev, and report back!

blancos-dev, 2024-11-22:
Hello @marcusmueller , thank you for your response. I am a telecommunications student and I'm learning about GNU Radio as part of my studies. I have downloaded and run the test file as indicated, but since I'm new to this, I'm not completely sure if the results I'm seeing are the expected ones.
Could you help me interpret if what I'm observing is normal? It would be very helpful to have a reference of what the correct result should look like so I can compare.
I appreciate your guidance while I learn to work with these OFDM systems.
[image]

bradleeharr, 2024-11-23:
@blancos-dev the bottom scope plot looks correct. The received data matches the transmitted data. 

Though, If I remember correctly, it looks correct because the same 50 bytes are transmitted every time, and the packet length is 50. So when a packet is dropped you wouldn't immediately see it in the output. 

Can you try changing the vector source from range(packet_len) to, for example, range(0, packet_len+14) (or just any other number than packet_len) to see if packets are still lost?

Wiktorijus, 2024-11-29:
Hi everyone, issue is still present in the current 3.10.10.0 GNU Radio version.
Anything less than packet_len is fine for any data source (random_source, file_source... ). Anything above packet_len and drops can be observed. There is and exception data_len = packet_len+packet_len//2 seems to be fine. 
Same issue can be observed in non-hierarchical OFDM flowgraph.

[image]
 
### System Information
OS: Windows 10
GR Installation Method: radioconda
### Specific Version
3.10.10.0

Wiktorijus, 2024-12-25:
After further testing it seems the issue lies with synchronization block (Schmidl & Cox OFDM synch.).

I think that for two reasons:
- first, if this block is bypassed or completely removed, and trigger tag is used on the Header/Payload Demux block for packet synchronization, simulation works with with much bigger number of samples with different data source. 
- For testing I used packet len of 10, vector source with range of 256 samples, but also random source with number of samples 5 000 000 and it worked perfectly. With synchronization block these parameters always result in drops of packets.
- secondly, based on the presentation from GRCon16, synchronization block (hier block) or more specifically plateau detector has a know bug. Long story short there is an edge case in which scheduler splits the plateau in a way, where two peaks are detected or **no peak is detected**, which would result in dropping of packets. 
GRCon16 - OFDM in GNU Radio, Matt La Pan  https://youtu.be/-DDgPL1UBUo?si=rNodEDq_lRlEHXzg&t=610

If this aforementioned bug was already fixed, please let me know. In any case synchronization block seems to be causing the issue, I'll try to found out more, possibly fix it, but if someone beats me to it, I'll be glad. In the video presenter mentioned that this bug was fixed but they might not have shared the fix, as they were using it for private purposes.

HirvaVekariya, 2026-11-11:
Comment for #6891
Hi all, we are a student team (ECE 310, Ahmedabad University) who measured this with GNU Radio 3.10.12.0. Each packet carried a sequence number, so we counted exactly which packets were lost (500 packets × 5 seeds, 50-byte packets).
Default TX (rolloff 0): ~4.1 % of packets lost with an ideal channel.
The losses are identical whether TX and RX run in one flowgraph or the RX runs afterwards on the stored TX signal, which supports @ffu's explanation (plateau timing + Header/Payload Demux), not a scheduler effect.
rolloff 1 has no effect, rolloff 2 → 99.8 %, rolloff 4 → 100 %; one zero sample between bursts → 100 %.
With AWGN, the default setting gets worse with more SNR (99.6 % at 15 dB, 94.4 % at 30 dB): noise randomises the plateau and partly hides the bug.
Since the existing QA only tests single packets, we would like to open a PR adding a multi-packet regression test and setting rolloff=4 in the example. Would that be welcome? Scripts and data: https://github.com/divyarajv1-ship-it/g1_gsoc_wifisensing

In M3 we logged the Schmidl & Cox trigger: with rolloff 0 it jitters between 69 and 73 samples, and a packet is lost exactly when its trigger arrives earlier than (frame length − 1) samples after the previous used trigger. This predicted every loss in 100 of 100 runs. With rolloff 4 the trigger position is fixed.

HirvaVekariya, 2026-11-11:
Opened #8271 with the regression test and example change.