gnuradio/gnuradio #6906: Selector oindex/iindex inputs report wrong type

Martyx00, 2023-10-10:
### What happened?

An attempt to set `oindex` for selector block fails with:
`thread_body_wrapper :error: ERROR thread[thread-per-block[3]: <block selector(6)>]: pmt_cdr: wrong_type 1`. (The number one is the index that I wanted to select)

The python code used to send the PDU:
`self.message_port_pub(pmt.intern(self.out_port_name), pmt.from_long(self.current_index))`

The same Python code was tested using `Message Debug` block and resulted in:
```
******* MESSAGE DEBUG PRINT ********
1
************************************
```

The flowgraph looks like this:
[image]

### System Information

OS: OpenSuse Tumbleweed
GR Installation Method: Native Distro package

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

Send PDU with `self.message_port_pub(pmt.intern(self.out_port_name), pmt.from_long(self.current_index))` to the `Selector` block where `self.current_index` is a valid integer in range between 0 and maximum number of outputs on the `Selector` block.

### Relevant log output

```shell
thread_body_wrapper :error: ERROR thread[thread-per-block[3]: <block selector(6)>]: pmt_cdr: wrong_type 1
```

willcode, 2023-10-10:
I just tested (with latest maint-3.10) a Message Strobe sending `pmt.from_long(1)` to a Selector's `oindex` port and a Message Debug `print`. Debug printed `1` and there was no complaint from the selector.

If you post the GRC, we might be able to try it out.

Martyx00, 2023-10-10:
Sure thing.
[selector_example.zip](https://github.com/gnuradio/gnuradio/files/12855451/selector_example.zip)

willcode, 2023-10-10:
OK, same result as yours using the example. Selector control message works directly from the Message Strobe but not from the Python code.

Martyx00, 2023-10-10:
From the initial assessment, is the problem in my code? I have used the suggestion from here: https://wiki.gnuradio.org/index.php/Selector

willcode, 2023-10-10:
Take a look at https://wiki.gnuradio.org/index.php/Message_Passing which shows that the post needs a `cons()`, i.e., should be
```
self.message_port_pub(pmt.intern(self.out_port_name), pmt.cons(pmt.PMT_NIL, pmt.from_long(self.current_index)))
```

closed 2023-10-10 by willcode