gnuradio/gnuradio #6914: Bug about QT GUI RANGE

ShiYaoDiving, 2023-10-14:
### What happened?

A bug happened!
if i drop a qt gui range widget into the grc, the whole application will crash with no information.

### System Information

OS:Windows10 LTSC X64
GR Installation Method:
radioconda-2023.07.26-Windows-x86_64.exe

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Open the GNU Radio Companion with "C:\Users\crackest\radioconda\python.exe C:\Users\crackest\radioconda\cwp.py C:\Users\crackest\radioconda C:\Users\crackest\radioconda\Scripts\gnuradio-companion.exe"
Search the QT GUI RANGE
drag it into the GRC.
the all crash....

### Relevant log output

```shell
no..
```

cafeclimber, 2023-10-14:
Nothing gets printed to the terminal where you launch GRC?

ShiYaoDiving, 2023-10-15:
It print the grc files that i load,and all seem work well until i drag a QT GUI RANGE widget.
And after the QT GUI RANGE drag , the widget will not show in the grc, the application and the terminal just closed with nothing print.
I try to run the GNURADIO Under the CMD Terminal, and after the drag, nothing printed , the application just exit.
I think it's not because of the script, because when i open the Windows Event Log, it shows the python.exe crashed.

argilo, 2023-11-19:
I made a few attempts, but I've been unable to reproduce this. Perhaps you could record a video of the failure occurring on your system, which might offer some clues as to what's going on.

ShiYaoDiving, 2023-11-21:
[https://res.cdn.office.net/assets/mail/file-icon/png/generic_16x16.png]2023-11-21 10-32-45.mkv<https://1drv.ms/u/s!AkH4QEdlyDKSjw3S7hvVd2CDUneo>
That is the video~~But i dont think that helps...cause it just crash without any clue....

argilo, 2023-11-21:
Thanks for the video. I didn't notice anything unusual in it.

However, I did notice that a long delay occurs for me the first time a QT GUI Range block is dropped onto the canvas, and this doesn't happen for other blocks. I checked with a debugger, and the delay is due to the `qtgui_python.cpython-310-x86_64-linux-gnu.so` shared library being loaded. That happens because of this import line:

https://github.com/gnuradio/gnuradio/blob/53de3dd32a9fc9ba1fa4e834487749340a6859a3/gr-qtgui/grc/qtgui_range.block.yml#L67-L68

The QT GUI Range block is the only block in GNU Radio 3.10 which does this. (On the main branch, the "QT GUI RFNoC Fosphor Display (Byte Vector)" block does the same.

I don't think there's any reason for the import, because the Qt GUI flowgraph template already includes `from gnuradio import qtgui`. I expect the following change would work, and get rid of the long delay (and hopefully also the crash):

```diff
diff --git a/gr-qtgui/grc/qtgui_range.block.yml b/gr-qtgui/grc/qtgui_range.block.yml
index 6642d204c4..085135abbf 100644
--- a/gr-qtgui/grc/qtgui_range.block.yml
+++ b/gr-qtgui/grc/qtgui_range.block.yml
@@ -64,8 +64,7 @@ asserts:
 - ${start <= stop}
 
 templates:
     imports: |- 
-        from gnuradio.qtgui import Range, RangeWidget
         from PyQt5 import QtCore
     var_make: self.${id} = ${id} = ${value}
     callbacks:
@@ -75,8 +74,8 @@ templates:
             win = 'self._%s_win'%id
             range = 'self._%s_range'%id
         %>\
-        ${range} = Range(${start}, ${stop}, ${step}, ${value}, ${min_len})
-        ${win} = RangeWidget(${range}, self.set_${id}, "${no_quotes(label,repr(id))}", "${widget}", ${rangeType}, ${orient})
+        ${range} = qtgui.Range(${start}, ${stop}, ${step}, ${value}, ${min_len})
+        ${win} = qtgui.RangeWidget(${range}, self.set_${id}, "${no_quotes(label,repr(id))}", "${widget}", ${rangeType}, ${orient})
         ${gui_hint() % win}
 
 documentation: |-
```

argilo, 2023-11-21:
Perhaps you could test this change by editing `C:\miniconda3\envs\gnuradio\Library\share\gnuradio\grc\blocks\qtgui_range.block.yml` and making the changes noted in the diff above (delete `from gnuradio.qtgui import Range, RangeWidget` and add `qtgui.` before `Range` and `RangeWidget`.)

ShiYaoDiving, 2023-11-21:
Thanks and it works!!!!! no delay and no crash now!!!

argilo, 2023-11-21:
Thanks for confirming that. Does the range slider work properly when you execute the flow graph?

ShiYaoDiving, 2023-11-22:
It works well!

argilo, 2023-11-22:
Thanks for confirming that. I opened #6970, so this should be fixed in the next GNU Radio release.

PR #6970 (2023-12-02): qtgui: Remove unnecessary imports from templates
## Description
When a QT GUI Range block is added to the canvas in GRC, there is a noticeable delay, which does not occur for other blocks. This happens because `from gnuradio.qtgui import Range, RangeWidget` causes a large number of shared libraries to be loaded. The import is unnecessary, because the flow graph template already imports `gnuradio.qtgui`:

https://github.com/gnuradio/gnuradio/blob/53de3dd32a9fc9ba1fa4e834487749340a6859a3/grc/core/generator/flow_graph.py.mako#L24-L30

The import can also cause a crash on Windows, as noted in #691 [... cut]
closed 2023-12-02 by willcode