gnuradio/gnuradio #6920: bus connections with variable names in them do not work

deeptho, 2023-10-16:
### What happened?

In gnuradio-companion:
When creating a block with "bus connections", this block only shows the correct number of outputs if
the connections are specified as numbers. If one of the numbers is replaced with a variable name, having the 
same value. an incorrect number of bus outputs is shown.

For example in a null source with 10 ouputs,
setting 'Bus connections" 
to [[0],[1], [*range(2,10)]]
works, but setting it to 
[[0],[ch1], [*range(2,10)]]
with ch1 being a variable with value 1
does not work

The attached example shows two null sources that should be identical but only the one
without a variable name in it works.

[test.zip](https://github.com/gnuradio/gnuradio/files/12921691/test.zip)

### System Information

OS:  Fedora 38
GR Installation Method: rpm
rpm -q gnuradio python3-gnuradio
gnuradio-3.10.4.0-6.fc38.x86_64
python3-gnuradio-3.10.4.0-6.fc38.x86_64

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.4.0

### Steps to Reproduce the Problem

Run gnuradio-companion on attached file and notice the difference between the two source blocks.

willcode, 2023-10-16:
Tagged as a feature request. There are a number of things that don't support variables in parameters, but things like this seem useful.

deeptho, 2023-10-17:
Why not add a comment to the documentation in the mean time?

Also: Currently the code silently fails: it just produces the wrong number of bus connections.

In other cases there is a stack trace (see older tickets here) when an accidental wrong input is
typed (e.g., forgetting a [  or ]). The various connections between other blocks are broken as well.  

Even trying "busify" a selector and then connecting it to a polyphase analyzer sometimes fails (complaining
about using an input 101 when only 100 are available).

So the user experience is not great....

Can you provide a pointer to the source file which handles the gui inputs?

willcode, 2023-10-17:
Here is some of the logic. There is also UI code to manage busses.
https://github.com/gnuradio/gnuradio/blob/main/grc/core/blocks/block.py

Busses are a lesser used (and developed) feature that could use some attention. I'm glad you're interested in looking at it!

deeptho, 2023-10-17:
I do not understand the code sufficiently to propose a good fix, but when running a debugger on a simple test.grc,
I noticed the foillowing:

In FlowGraph.py, at some point the  function "evaluate" on like 301 is called with
``expr='[[0],[ch1], [*range(2,10)]]```
and namespace and local_namespace both set to None.

It then tests for "namespace is not None" and as a result, the code calls 
self.eval_cache.set_default(...) .

I think the bug is that "namespace is None" just means "use the default namespace"
but because of the test for "is not None", eval is never even tried. 

I have verified that with the following debugging code added, gnuradio-companion shows up
with a properly connected flowgraph:
```

    def evaluate(self, expr, namespace=None, local_namespace=None):
        """
        Evaluate the expression.
        """
        # Evaluate
        if not expr:
            raise Exception('Cannot evaluate empty statement.')
        if expr =='[[0],[ch1], [*range(2,10)]]' or namespace is not None:
            try:
                return eval(expr, namespace, local_namespace)
            except Exception as e:
                return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace))
        elif namespace is not None:
            return eval(expr, namespace, local_namespace)
        else:
            return self._eval_cache.setdefault(expr, eval(expr, self.namespace, local_namespace))
```
As you can see, the code tries first a regular eval, and if it fails (because variable ch1 is not
yet defined), then -- and only then -- it falls back to "setdefault".
For some reason the setdefault is essential

Debugging also shows that in some calls which have
expr =='[[0],[ch1], [*range(2,10)]]' ,
the caller calls "evaluate: with namespace set to self.namespace

However, in param.py the class "Param" does not have a self.namespace,
so it is not possible to pass it as value for namespace. Also, my "debug" code
shows that self.namespace is also not even needed

I am not sure what the correct fix is but perhaps this helps to suggest a solution?