gnuradio/gnuradio #6921: Flowgraph containing RFNoC FFT block cannot start when FFT length is 1024

lucab0212, 2023-10-18:
### What happened?

Please take note of the additional debug statements inserted into the UHD source files in "System Information". This bug is related to the [UHD RFNoC property resolution process](https://files.ettus.com/manual/page_properties.html). Take note of the sections about property resolution, atomic_item_size and spp.

In GNU Radio it is not possible to use an RFNoC FFT block with FFT lengths of 1024 (or higher). It causes the debug output mentioned below.
However, FFT lengths of 1024 are possible when directly using UHD's Python API: The example script [rfnoc_rx_to_file](https://github.com/EttusResearch/uhd/blob/master/host/examples/rfnoc_rx_to_file.cpp) implements a very simple flowgraph that blocks (with customizable parameters) can be inserted to. Calling the script
```
./rfnoc_rx_to_file --spp=256 --block-id FFT --block-props length=1024
```
executes successfully, the debug output being
```
[DEBUG] spp = 18446744073709551615
[DEBUG] ais.get() = 1

[DEBUG] spp = 2000
[DEBUG] ais.get() = 1

[DEBUG] spp = 2000
[DEBUG] ais.get() = 1024

[DEBUG] spp = 1024
[DEBUG] ais.get() = 1024
```
This is because in the script, the FFT length property is set *after* commiting the flowgraph meaning that the property resolver also gets active after commiting the flowgraph. Note this comment in the script:
```
/************************************************************************
     * Set up sampling rate and (optional) user block properties. We do this
     * after commit() so we can use the property propagation.
     ***********************************************************************/
```
I created my own version of this example script where the FFT length property is set *before* commiting the flowgraph and it results in the same error as in my GNU Radio flowgraph, producing the same debug output mentioned below.

This problem should be solved by making GNU Radio first commit the RFNoC graph it implements and then set the parameters. This is however not possible by changing parameters in the GNU Radio flowgraph.

A current workaround is by editing [uhd/rfnoc_fft_block_control.cpp](https://github.com/EttusResearch/uhd/blob/master/host/lib/rfnoc/fft_block_control.cpp) and setting `DEFAULT_LENGTH` to 1024 because then the property resolver doesn't have to be called. This however makes the use of any other FFT length impossible.

### System Information

OS: Debian 12
GR Installation: 3.10.7 (from source)
UHD Installation: 4.4.0.0 (from source, with manual edits mentioned below)
USRP: N310 (this bug should be universal to all USRPs that support RFNoC)

Manual debug statements inserted into [the atomic item size property resolver in uhd/.../rfnoc_rx_streamer.cpp](https://github.com/EttusResearch/uhd/blob/471af98f6b595f5fd52d62303287d968ed2a8d0b/host/lib/rfnoc/rfnoc_rx_streamer.cpp#L243):

```cpp
//...
if (ais.is_valid()) {
    const auto spp = this->rx_streamer_impl::get_max_num_samps();
    std::cout << "\n[DEBUG] spp = " << spp;
    std::cout << "\n[DEBUG] ais.get() = " << ais.get();
               
    if (spp*4 < ais.get()) {
    //...
```
Additionally in the second if case above, you have to multiply `spp` by four: `spp*4 < ais.get()`. This is an error in the UHD source code. `spp` has to be multiplied by four because `spp` has units of samples whereas `ais.get()` has units of bytes. This bug was found by @rkossler

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

- load a bitfile containing the RFNoC FFT block onto a USRP that supports RFNoC
- create a GNU Radio flowgraph that uses the RFNoC FFT Block, e.g.:
RFNoC RX Radio -> RFNoC FFT -> RFNoC Rx Streamer -> Complex to Mag -> Log10 -> QT GUI Vector Sink

- set the FFT Length in the FFT block to 1024
- try starting the flowgraph

### Relevant log output

```shell
[INFO] [0/LogPwr#0] Setting default MTU forward policy.

[DEBUG] spp = 512
[DEBUG] ais.get() = 1

[DEBUG] spp = 512
[DEBUG] ais.get() = 1
 
[DEBUG] spp = 512
[DEBUG] ais.get() = 4096

RuntimeError: ValueError: samples per package must not be smaller than atomic item size
```

marcusmueller, 2024-02-14:
This is a bug, and it does surface in GR usage, but far as I can tell, it needs to be solved within UHD; @mbr0wn I couldn't pinpoint whether this is fixed in upstream UHD, and being vivado license-constrained I've not build test FPGA images, so unless you disagree, I'd close this?

mbr0wn, 2025-02-03:
The most recent version of UHD has lots of updates to the FFT block, and resolves this issue.

closed 2025-02-03 by mbr0wn