gnuradio/gnuradio #6923: grc generates wrong python code when using realtime scheduling

cxp10, 2023-10-21:
### What happened?

If a flowgraph project enables realtime scheduling in Options, grc will generate:
```
def main(top_block_cls=transceiver_CSS_loopback, options=None):   
    if gr.enable_realtime_scheduling() != gr.RT_OK:   
        gr.logger("realtime").warning("Error: failed to enable real-time scheduling.")
```
which will result in
`AttributeError: 'gnuradio.gr.gr_python.logger' object has no attribute 'warning'`

This bug is probably caused by a typo of `gr.logger.warn()` 
This bug is similar with a closed issue  https://github.com/gnuradio/gnuradio/issues/1924.

### System Information

OS: Linux Ubuntu 22
GR Installation Method:  Pybombs

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.8.0-rc2

### Steps to Reproduce the Problem

Executing a flowgraph with realtime scheduling enabled in gnuradio-companion 3.10

willcode, 2023-10-21:
Replicated.

@marcusmueller could you take a look at this? We have a few calls to `warning()` in GRC and modtool. This is the only one that has an external effect.

marcusmueller, 2023-10-22:
@willcode  will have a look, thanks

marcusmueller, 2023-10-22:
Hi @cxp10, sorry for the bug (I'm the one who caused it).

Luckily, we can "work around" the bug by actually fixing the underlying issue, namely that you don't give programs the ability to ask for realtime scheduling (which you *really* want if you're doing real-time signal processing). 

It's rather easy: run

```shell
echo "$(whoami)  -  rtprio  99" | sudo tee /etc/security/limits.d/99-rtprio.conf
```

will tell the system that next time you log in, your user's tasks gain the ability to ask for real-time scheduling.

marcusmueller, 2023-10-22:
@willcode all the other calls to `warning` are on `import logging` loggers, so do not apply here.

cxp10, 2023-10-22:
Thanks for the advice, and the bug fixing! 
I just decided to leave this issue open till request merged.

marcusmueller, 2023-10-22:
That's 100% appropriate! The issue gets closed when the fix is in the tree, no earlier.

PR #6924 (2023-10-24): GRC: generated python code use gr.logger.warn instead of ….warning
This only triggered when real-time scheduling could not be enabled

## Description

Due to a mistake, we used `gr.logger().warning` instead of `gr.logger.warn`.

## Related Issue

Fixes #6923

## Which blocks/areas does this affect?

GRC code gen

## Testing Done
closed 2023-10-24 by willcode