gnuradio/gnuradio #6927: 3.10.8.0 build failure

chenrui333, 2023-10-24:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

<details>
<summary>error build log</summary>

```
  /opt/homebrew/include/spdlog/fmt/fmt.h:27:14: fatal error: 'spdlog/fmt/bundled/core.h' file not found
  #    include <spdlog/fmt/bundled/core.h>
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  1 error generated.
```

</details>

full build log, https://github.com/Homebrew/homebrew-core/actions/runs/6589047383/job/17902765159
relates to Homebrew/homebrew-core#151928

willcode, 2023-10-24:
Homebrew fails with

/opt/homebrew/include/spdlog/fmt/fmt.h:27:14: fatal error: 'spdlog/fmt/bundled/core.h' file not found

marcusmueller, 2023-10-24:
Ah I think I know what's happening. In io_signature.h, we

    #include <spdlog/fmt/fmt.h>

which is the right thing to do – that internally includes either the bundled fmt or the external, depending on how the system's spdlog is configured in its tweakme.h, where `SPDLOG_FMT_EXTERNAL` gets set (or not).

Now, other spdlog-bundled headers in  spdlog/fmt/ (for example, spdlog/fmt/bin_to_hex.h) do `#include <spdlog/common.h>`, which ensures that tweakme.h is included and so `SPDLOG_FMT_EXTERNAL` is correct.

I'm trying to upstream a fix. 

In the meantime, I'll work around this in GNU Radio, please stand by.

closed 2023-11-17 by willcode

marcusmueller, 2024-01-03:
@dpward  sadly, we have in the past encountered multiple systems where we couldn't rely on spdlogConfig.cmake, so that's sadly not an option in the current environment.

PR #6928 (2023-11-17): spdlog: include spdlog system settings header before including spdlog/fmt/
Otherwise, spdlog/fmt/anything.h might include the internal
fmt/anything.h, although spdlog is set up to use an external fmt.

This is addressed upstream in
https://github.com/gabime/spdlog/issues/2922 .

## Description

Just an `#include <spdlog/tweakme.h>` with an explanatory comment on why reordering the include would be a bad idea.

## Related Issue

Hopefully:

Fixes #6927

## Which blocks/areas does this affect?

a couple, sadly including public ru