gnuradio/gnuradio #6929: Rename all instance of fosphor to RFNoC fosphor

smunaut, 2023-10-24:
What was imported in tree is _not_ fosphor which is a distinct project, it's RFNoC fosphor and it should be clearly marked as such in the documentation, the ui and the code.

willcode, 2023-10-24:
@mbr0wn I agree it's confusing to users to have a fosphor that's not fosphor and doesn't operate the same way. This hasn't been backported yet, so a name change is not an issue.

smunaut, 2023-10-24:
Yeah, I should have raised it earlier, but at least it's before a stable release is done.
I can have a shot at it, but not sure when the next release is scheduled for ?

willcode, 2023-10-24:
We were hoping to have a formatter block that made the RFNoC Fosphor usable from a stream, but making that as flexible as the other GUI sinks (e.g., user-defined scaling) got bogged down. We may backport the display block only for the Jan 2024 release so it's at least usable in RFNoC graphs.

smunaut, 2023-10-24:
Ok, I can definitely get the display block renaming done by Jan 2024.
I decided not to go to CCC this year and instead use the week to work on fosphor (complete its long overdue transition to GL compute instead of OpenCL so it's way easier to install). I can add the renaming on my todo list.

willcode, 2023-10-24:
GL would be great.

willcode, 2023-10-24:
Any interest in Fosphor in-tree? We'd already have GL as an optional dep to enable the RFNoC Fosphor. Are there other build time or runtime deps for Fosphor?

mbr0wn, 2023-10-25:
Hey @smunaut and @willcode, I agree and I can also do the module renaming myself (being the one who upstreamed it). I should have discussed this with you beforehand; as you can see I mostly auto-renamed blocks from gr::ettus to gr::uhd namespace, hence the naming.

mbr0wn, 2023-10-26:
@smunaut @willcode Suggestion: How about I rename the source code files that are currently called `fosphor` to `rfnocf15*` (e.g., `fosphor_display.h` -> `rfnocf15_display.h`). The 'f15' abbreviation for fosphor is something the Sylvain used for the source code of the FPGA code. This avoids the filenames and class names being too long (`rfnoc_fosphor_display.h`), and it further reduces similarity between the classic GPU-based fosphor. I would keep the labels and description strings as "RFNoC Fosphor" like @smunaut suggested, just so people don't have to rack their brains what f15 could mean.

smunaut, 2023-10-26:
Sure sounds good to me. I might do `rfnoc_f15` because I must admit at first I didn't see the "f" but other that that, I think it's a good way to keep names shorter.

closed 2023-11-08 by willcode