gnuradio/gnuradio #6940: Python basic_block -- consume_items question

SaikWolf, 2023-11-02:
### Feature Description

At the moment just a question to better understand this chunk of code.

[Python basic_block gateway.py](https://github.com/gnuradio/gnuradio/blob/main/gnuradio-runtime/python/gnuradio/gr/gateway.py)
```python
class basic_block(gateway_block):
    """
    Args:
    name (str): block name

    in_sig (gr.py_io_signature): input port signature

    out_sig (gr.py_io_signature): output port signature

    For backward compatibility, a sequence of numpy type names is also
    accepted as an io signature.
    """

    def __init__(self, name, in_sig, out_sig):
        gateway_block.__init__(self,
                               name=name,
                               in_sig=in_sig,
                               out_sig=out_sig,
                               block_type=gr.GW_BLOCK_GENERAL
                               )

    def consume_items(self, nitems):
        pass
```

Is there a reason for the `consume_items` function to be undefined for this block?

I guess would it make better user interface to have it throw an `NotImplementedError` if this is intentional?

### Feature Urgency

low (just an idea)

marcusmueller, 2023-11-02:
Issue tracker, not a question / answer platform

marcusmueller, 2023-11-02:
(sorry that we're a bit strict on that, but we do get too many issues that aren't really bug reports, nor feature requests. Our mailing list  and our matrix chat are always open for questions! Please keep the bug tracker clean.)

closed 2023-11-02 by marcusmueller