gnuradio/gnuradio #6945: Flowgraph id of `gnuradio` results in Python import errors

ryanvolz, 2023-11-07:
### What happened?

If you set your flowgraph id to `gnuradio`, then when GRC compiles it the Python file will be `gnuradio.py` which masks the actual `gnuradio` Python module resulting in a somewhat cryptic error message:
```
Generating: 'C:\\Users\\rhofmann\\Documents\\gnuradio.py'

Executing: C:\Users\rhofmann\radioconda\python.exe -u C:\Users\rhofmann\Documents\gnuradio.py

Traceback (most recent call last):
  File "C:\Users\rhofmann\Documents\gnuradio.py", line 13, in <module>
    from gnuradio import qtgui
  File "C:\Users\rhofmann\Documents\gnuradio.py", line 13, in <module>
    from gnuradio import qtgui
ImportError: cannot import name 'qtgui' from partially initialized module 'gnuradio' (most likely due to a circular import) (C:\Users\rhofmann\Documents\gnuradio.py)

>>> Done (return code 1)
```

This came up in a bug report to radioconda: https://github.com/ryanvolz/radioconda/issues/75.

I know that some flowgraph names are disallowed, and perhaps it's too much work to make the list exhaustive (e.g. you wouldn't want to name a flowgraph `numpy` either for similar reasons...), but maybe `gnuradio` deserves to be disallowed as well?

### System Information

OS: Windows
GR Installation Method: radioconda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.7.0

### Steps to Reproduce the Problem

Set a flowgraph id to `gnuradio` and try to run it. Moreover, once you've done that, you can't run any flowgraphs from that directory until you delete `gnuradio.py`.

willcode, 2023-11-07:
Would this be true of any visible module?

ryanvolz, 2023-11-07:
> Would this be true of any visible module?

Yes, I think so.

willcode, 2023-11-07:
There could be a couple special cases that we block. Someone will undoubtedly then use an OOT module name or some GR dep (like you mention, numpy) or even library modules. Not sure how much we'd have to do to make this generally useful.

ryanvolz, 2023-11-07:
Maybe it's enough to have the bug report as something searchable that can point people in the right direction of changing their flowgraph ID.

PR #6962 (2023-11-16): grc: backlist gnuradio as valid id

## Description

See #6945
## Related Issue

Fixes #6945
## Which blocks/areas does this affect?

## Testing Done

## Checklist
commit 03fb0dcf (2023-11-16): grc: backlist gnuradio as valid id

Fixes #6945
commit 4f95ac6c (2023-11-29): grc: backlist gnuradio as valid id

Fixes #6945

(cherry picked from commit 03fb0dcf8a006a333048d481a7bb6ccdf5bcf155)
closed 2023-11-16 by willcode