gnuradio/gnuradio #6951: Windows builds don't accept ints in `pmt.from_long`

marcusmueller, 2023-11-12:
### What happened?

[as found while writing a unit test](https://github.com/gnuradio/gnuradio/pull/6950#issuecomment-1807279633), the windows CI [fails with the following](https://github.com/marcusmueller/gnuradio/actions/runs/6843384944/job/18605928276#step:6:9194):

```
   File "%SRC_DIR%\gr-blocks\python\blocks\qa_msg_pair_to_var.py", line 38, in test_calling
    test_msg = pmt.cons(pmt.intern("foo"), pmt.from_long(canary))
                                           ^^^^^^^^^^^^^^^^^^^^^
TypeError: from_long(): incompatible function arguments. The following argument types are supported:
    1. (x: int) -> pmt.pmt_python.pmt_base

Invoked with: 3736059631
```

uuuuuuuh, pretty sure 3736059631==0xdeadbeef is an integer value, and especially a valid python `int`. sure, it's larger than `1<<31`, but why is this failing here… will have to dig up what the correct behaviour is and why we're failing it (and whether PMT is missing a kind of obvious test case).

### System Information

OS: Windows Conda CI build
GR Installation Method: cönda

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

https://github.com/gnuradio/gnuradio/pull/6950/commits/20da7125373d4336a2024c8d868f2090c9d6c66c

### Steps to Reproduce the Problem

run the CI when trying to `canary = 0xdeadbeef; pmt.from_long(canary)`

### Relevant log output

```shell
test_msg = pmt.cons(pmt.intern("foo"), pmt.from_long(canary))
                                           ^^^^^^^^^^^^^^^^^^^^^
TypeError: from_long(): incompatible function arguments. The following argument types are supported:
    1. (x: int) -> pmt.pmt_python.pmt_base

Invoked with: 3736059631
```

marcusmueller, 2023-11-12:
I bet it's pybind being smart about `from_long` being the least sane API for integer conversion, but, who knows. @jsallay knows, maybe.

jsallay, 2023-11-13:
Off the top of my head, my guess is that it had to do with a long being 32
bits on Windows and 64 bits on 64 bit Linux.

My guess is that if you picked a smaller value that it would work just
fine. If pybind has a range check, then I could see it failing.