gnuradio/gnuradio #6954: peak identification

joe1365, 2023-11-14:
### Feature Description

there is a requirement of peak identification for identifying active signals for making cnn based radio fingerprinting in a 20 MHz instantaneous bandwidth

### Feature Urgency

high (impacts broad portion of ecosystem)

### More Information

to design a radio fingerprinting software it is essential to identify the peaks to select the sample signals for feature matching.

willcode, 2023-11-14:
If someone writes a general purpose multi-peak finder, it could be an interesting addition. Until then, you could use numpy peak detection in a custom python block, using an FFT as input.

We can leave this open in case anyone is interested in contributing a flexible, high-performance, multi-peak frequency detector.

joe1365, 2023-11-14:
Thank you jeff really thankful to see the promt and timely help from your
end.

I have made some coding towards a general purpose peak finder in hf and vhf
range. Hopeful to finish the testing soon.

Warm regards

luzpaz, 2023-12-13:
@joe1365 is there a branch that you can share that you're developing it on ?

marcusmueller, 2023-12-13:
note that "we" (as in: the CEL, my former institute) used to have gr-specest, but I haven't maintained it in the last couple of years.

Specest does contain numerous general purpose spectrum and pseudospectrum estimators, including "peak-finding ones" that are superresolution, i.e., don't suffer the frequency resolution problem that you get with an FFT-based approach, where only the frequencies that fall onto the raster given by $n\frac{f_{\text{sample}}}{N}, \quad n=0,\ldots, N-1 $, $N$ being the length of the FFT, get detected perfectly and other suffer a resolution and/or SNR penalty.

willcode, 2023-12-13:
@luzpaz do you have a use case in mind?

joe1365, 2023-12-14:
Yes we are working on it and have achieved some progress too to identify
peaks in frequency hopping radios