gnuradio/gnuradio #6957: feature-grc-qt Minor Issues

dkozel, 2023-11-15:
### What happened?

This is a collection of minor issues or notes found while testing the new Qt version of GRC.

Last checked on 4e444b7ff

- [ ] Properties menu does not open when using a block's right click menu
- [ ] Opening a GRC file immediately modifies it so it cannot be closed without saving
- [ ] Right clicking on an unselected block does not select it and give access to options like enable, disable, properties

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

v3.11.0.0git-785-g4e444b7ff

### Steps to Reproduce the Problem

ss

haakov, 2023-11-15:
- [ ] Ctrl+Scroll zooms in on the top left corner. Maybe it should zoom in on where the mouse is?

AsriFox, 2023-11-30:
> * [ ] Running on Wayland: `qt.qpa.wayland: Wayland does not support QWindow::requestActivate()`

@haakov Where exactly does this occur? I can't reproduce this in GNOME Wayland (Ubuntu 22.04LTS VM), Plasma Wayland (Ubuntu 23.10), Hyprland, cage or weston (Arch).

AsriFox, 2023-12-04:
I can reproduce the `qt.qpa.wayland` bug in weston (Arch), GNOME 42.9 (Ubuntu 22.04LTS VM) and GNOME 45.2 (Fedora 39 VM). However, I can't reproduce it in Hyprland, cage (Arch) or Plasma Wayland (Ubuntu 23.10). I can see the message in GNOME 45 (Ubuntu 23.10), but menus appear in correct locations.

AsriFox, 2023-12-04:
The problem with `add_block_from_query` looks like this:
[image]
Here are the test logs from running `ctest -R grc_tests` with `QT_QPA_PLATFORM=wayland` and `QT_QPA_PLATFORM=xcb` on different systems:
[Arch_Hyprland_wayland_Test.log](https://github.com/gnuradio/gnuradio/files/13550657/Arch_Hyprland_wayland_Test.log)
[Arch_Hyprland_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550658/Arch_Hyprland_xcb_Test.log)
[Fedora_Gnome45.2_wayland_Test.log](https://github.com/gnuradio/gnuradio/files/13550659/Fedora_Gnome45.2_wayland_Test.log)
[Fedora_Gnome45.2_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550660/Fedora_Gnome45.2_xcb_Test.log)
[Ubuntu_Gnome42.9_wayland_Test.log](https://github.com/gnuradio/gnuradio/files/13550661/Ubuntu_Gnome42.9_wayland_Test.log)
[Ubuntu_Gnome42.9_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550663/Ubuntu_Gnome42.9_xcb_Test.log)
[Ubuntu_Gnome45_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550664/Ubuntu_Gnome45_xcb_Test.log)
[Ubuntu_Plasma_xcb_Test.log](https://github.com/gnuradio/gnuradio/files/13550665/Ubuntu_Plasma_xcb_Test.log)

dkozel, 2024-02-06:
Thanks for tackling many of the initially found minor issues. Here's another set I found while looking at the new features. Again, just notes so we don't forget them over time and anyone interested has a list to look at.

- [ ] Block comments only display as much text as fits exactly under the block's outline. Replicate by inserting a long comment.
- [ ] Block comments only display one line of text and block property is a single line field rather than multi line.
- [ ] Note block displays text incorrectly. Likely a due to one or both of the above items. Open mpsk_stage6.grc example.
- [ ] Blocks rotated 180 degrees are visually upside-down. Gtk GRC flips the block ports but not text in this case, easier to read.
- [ ] Connector arrows face in the wrong direction on input ports on a rotated block.
- [ ] Block property dialog fields all display "TYPE" rather than actual parameter type.
- [ ] Variable Editor window shows "<Open Properties>" for many variable values rather than a calculated value. For example a `QT GUI Range` block's variable.
- [ ] Connector curves don't render well when it's start and end are too close together.
- [ ] Align blocks feature not yet implemented.

dkozel, 2024-02-10:
- [ ] Python Block properties UI not implemented. Currently gives a raw text field for the code rather than an external editor.
- [ ] Virtual Sink ports do not change color based on connected port types
- [ ] Wildcard and Bus Connection ports are white and need an outline to be visible in a light theme

haakov, 2024-02-10:
>     * [ ]  Zooming in on the flowgraph canvas works using the `+` on the number pad but not on the main `+` key, need to add '=' as an alias, at least for US keyboard. in the Gtk codebase: [e50c3f8](https://github.com/gnuradio/gnuradio/commit/e50c3f80dd68c29f0f1331692dedefb95ab46ca7)

Should be fixed now for the Zoom in action. Is this just for zooming out, or is it for zooming out as well?

lukazlim, 2024-03-07:
I have a related issue to report for the feature-grc-qt branch too, last checked on https://github.com/gnuradio/gnuradio/commit/9f50457ddce76b1a30d94dfb6e29d6d5873fb436

- [ ] After removing the last-opened flowgraph, a FileNotFoundError error is returned when starting up with `gnuradio-companion --qt`
- [ ] After removing the last-opened flowgraph, a segfault occurs when starting up with `gnuradio-companion`

The `--gtk` version has no issues, it defaults to a new flowgraph which should be the intended behaviour.

[image]

Restoring/moving the flowgraph back to its original directory will stop those errors until the program is closed while on a new unsaved flowgraph.

haakov, 2024-03-24:
>     * [ ]  After removing the last-opened flowgraph, a segfault occurs when starting up with `gnuradio-companion`

It looks like GRC Gtk is segfaulting, not GRC Qt. Can you confirm this? In that case, this should probably be a separate Github issue.

marcusmueller, 2024-12-26:
Hi. I admire the intent of keeping these things together in a list as "meta-issue", but we can't track progress on this, nor hint at any issues to interested outsiders, nor is it sensible to keep N different bug discussions alive.

Instead, I created a new issue for every outstanding `[ ]` in this list. Please refer to these issues.

Another remark: if we find ourselves in need of keeping track of progress across multiple issues, we can do that: github has "projects". I have, as an example, and as something well-tackleable in e.g. a hackfest, created the "[GRC-Qt feature parity](https://github.com/orgs/gnuradio/projects/10/views/2)" project.

I'll close this meta issue (this doesn't imply any of the issues mention within are closed), and would love for people to continue this excellent work by adding a quick, if wanted, completely formless issue, tag it "GRC-Qt" and whatever else is appropriate, and add it to the GRC-Qt project (we've had that for a long time!) and/or the aforementioned feature parity project. Thanks!

closed 2024-12-26 by marcusmueller

haakov, 2024-12-26:
Good point. Thanks for dividing them up!

PR #6976 (2023-12-23): grcqt: Minor Issues Fixes
## Description
Fixes for minor issues with the Qt version of GRC:
- [x] Ctrl+Scroll zooms in on the top left corner. Maybe it should zoom in on where the mouse is?
- [x] Zooming with Ctrl +, Ctrl -, Ctrl 0 is not implemented
- [x] Generate fails if the flowgraph hasn't been saved
- [x] Right clicking on an unselected block does not select it and give access to options like enable, disable, properties

Note: "Zoom in" shortcut is set to Ctrl =, because Ctrl + requires Shift to be pressed.

## Related Issue
Fixes #6957 

## Which blocks/areas does this affect?
GRC: Qt G [... cut]