gnuradio/gnuradio #6958: Compatibility problems between Pluto SDR Sink/Source (gr-iio) and Soapy Pluto Sink/Source (soapy)

vicruz99, 2023-11-15:
### What happened?

Using the blocks Pluto SDR Sink and Pluto SDR Source (from the module gr-iio), I was able to send and receive a signal in the same Adalm-Pluto with no problems.

Then, because I wanted to use the Soapy library, I installed the Soapy Pluto plugin (https://github.com/pothosware/SoapyPlutoSDR) and then used the blocks Soapy Pluto Sink and Soapy Pluto Source to again send the same signal, which went fine.

Then, when I tried to run the initial blocks again, Pluto SRD Sink and Pluto SDR Source, despite the program starting and being apparently able to connect to the Plutos, I wasn't receiving any signal, or being able to control on run-time the transmitting or receiving power of Adalm-Pluto.

When I tried to use another Pluto, I could send signals again. However, as soon as I use the Soapy blocks, the blocks from the gr-iio module stop working. I think there may exist a compatibility issue between these two libraries. Or maybe the Soapy blocks alter some configurations in the Plutos which render the gr-iio blocks useless.

### System Information

OS: Fedora 39 on Virtual Box
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0

### Steps to Reproduce the Problem

- Send and receive a signal using blocks Pluto SDR Sink and Pluto SDR Source. Everything should go well.
- Send and receive a signal using blocks Soapy Pluto Sink and Soapy Pluto SDR. Everything should go well.
- Send and receive a signal again using blocks Pluto SDR Sink and Pluto SDR Source. You should be able to start the flowgraph, but not of receiving any signal.

willcode, 2023-11-15:
This sounds like the Soapy PLUTO blocks are keeping the device open. Try quitting GRC after each run and see if things work. This is likely a Soapy driver issue, not a GNU Radio issue, but we can leave this open for now.

vicruz99, 2023-11-15:
If i just quit GRC, the error persists. However, if I unplug the device (leaving without power) and then connect it again, the Pluto SDR Sink and Pluto SDR Source now work again, which leads me to believe your explanation, which states that this is a Soapy driver issue is possibly correct.

willcode, 2023-11-15:
If quitting GRC has no effect, it's not even the Soapy drivers holding something open. The drivers may be leaving something in a bad state. In any case, not a GNU Radio issue, so I'll close it here.

closed 2023-11-15 by willcode