gnuradio/gnuradio #6967: Some PMT Python bindings are broken

argilo, 2023-11-19:
While working on #6966, I noticed that several PMT Python bindings appear non-functional, printing the following error when invoked:

```
Did you forget to `#include <pybind11/stl.h>`? Or <pybind11/complex.h>,
<pybind11/functional.h>, <pybind11/chrono.h>, etc. Some automatic
conversions are optional and require extra headers to be included
when compiling your pybind11 module.
```

The affected functions are:

* `pmt.make_any`
* `pmt.any_ref`
* `pmt.any_set`
* `pmt.map`
* `pmt.read`
* `pmt.write`
* `pmt.serialize`
* `pmt.deserialize`

I performed this testing on the `main` branch.

When testing with GNU Radio 3.10, the first three functions do not print an error because they use `boost::any` instead of `std::any`, but I'm not sure whether it's possible to pass in a `boost::any` argument.

themox, 2023-11-20:
Experiencing same issue; haven't tested other functions but can confirm exact behavior with pmt.serialize and pmt.deserialize.  

Using Python 3.10.12, GNURadio 3.10.7.0.2, installed via Brew on Mac

myyxl, 2023-12-13:
I encounter the same issue while using `serialize` and `deserialize`.

Python 3.11.4, GNURadio 3.10.5.1 and installed using the package manager on Ubuntu.

willcode, 2023-12-13:
serialize and deserialize take a streambuf arg, which is also apparently not mapped to the wrappers.

willcode, 2023-12-13:
@mormj @jacobagilbert might have ideas here.

marcusmueller, 2024-02-04:
wrapping std::any-consuming functions to Python isn't very sensible; so these should be removed.

PR #7111 (2024-02-22): PMT: remove wrapping of C++-specific types
- PMT: there's no std::any in Python, don't pybind make_any, any_ref, _set
- PMT: don_t wrap C++ pmt_t(pmt_t) functor-consumping PMT map to Python
- PMT: don_t wrap C++ iostreams, std::streambufs to Python

## Description

C++ std::any has no correspondent type Python, so we shouldn't wrap it

Same for iostreams, streambuffers, and C functions of std::shared_ptr<pmt_base> to shared_ptr<pmt_base>

## Related Issue

Fixes #6967

## Which blocks/areas does this affect?

PMT wrappings that
closed 2024-02-22 by willcode